MatteCarra/AccA #1: Voltage control doesn't convert uV to mV for proper display of voltage limit

Unspec7, 2019-02-28:
Steps to reproduce:
1. Open AccA
2. Edit Config
3. Edit voltage limit
4. Enable voltage limit
5. Enter valid limit (e.g 4100)
6. Click Ok, save config
7. Go back into config editor and edit voltage limit
8. Voltage limit is now displayed in microVolts (4100 -> 4100000)

Acc reports voltage limits in microVolts, not milliVolts even though the voltage limit is set via milliVolts, which is probably why the voltage limit is displaying incorrectly

MatteCarra, 2019-02-28:
Thanks for the report.
My phone does not support voltage control so I was not able to test it.
VR25 is working on making every voltage expressed in mV so this problem will be fixed in the next acc release, but I might fix it for the app before

MatteCarra, 2019-02-28:
Let me know if it works for you.
The apk is zipped:
[app-release.zip](https://github.com/MatteCarra/AccA/files/2917161/app-release.zip)

closed 2019-02-28 by MatteCarra