MatteCarra/AccA #9: Battery level displayed in ACCA of by about 3-5%

undermark5, 2019-03-08:
I'm running Google pixel 3 with latest versions of the app and Acc installed. Updated app potential cause, but not sure.

VR-25, 2019-03-08:
There's a debate on whether we should show the true (hardware) values over software-masked values. We're currently using the first.

acc -i is more accurate than the system in this regard. Thus it's safe to say that system is of by 3-5%, not the battery management subsystem itself.

undermark5, 2019-03-08:
That makes sense.
Though it is somewhat confusing when I've got it set to stop charging at 85% and it stops charging at about 80% (as shown by the battery indicator).
Perhaps if there is a debate over what should be shown/used, it would make sense to provide a choice to the user if possible.

VR-25, 2019-03-08:
The `dumpsys battery` command doesn't always work. Thus, using values provided by the system would be a curse, rather than a potential solution.

My idea is implementing a capacity tolerance feature (e.g., capTolerance=+5). What are your thoughts on that?

undermark5, 2019-03-11:
Should work out just fine I think.

MatteCarra, 2019-05-11:
Check out the new acc config file.
Your issue should be now solved!

closed 2019-05-11 by MatteCarra