MatteCarra/AccA #22: Some questions and suggestions

friendlytherapist, 2019-05-18:
How do I update the app without loosing my data?
Can you add an in app update checker?
Can you create a default profile for each phone that is the manufacturers settings, as in what it would charge to if I completely removed ACC. Or a default check box next to each slider.

MatteCarra, 2019-05-21:
Sorry for the late response.

A update notification mechanism should be already included.

The second idea is good.

ChingghisKhan, 2019-07-21:
Hi, I am having issues installing ACCA:
Installation in Magisk seems corrupted.
Magisk tells me an update is availbale but installation fails.
Terminal tells me no update availble.
Please see screenshots.

Thanks
[image]
[image]
[image]
[image]
[image]

VR-25, 2019-07-21:
@ChingghisKhan, the upgrade command and back-end installer for AccA had issues. The latest updates fix these. ACC is already up. The app will follow soon. Once you install the new AccA, `acc --upgrade` will work as expected.

Note: the ACC Magisk module is uninstalled by AccA. If you see an ACC zombie in _Magisk Manager > Modules_, mark it for removal.

ChingghisKhan, 2019-07-22:
Hi,

 

Thanks for your fast help, it works like charm now J

 

Best regards from Germany

 

Robert

closed 2019-08-10 by MatteCarra