MatteCarra/AccA #26: App not working

bridges86406, 2019-06-02:
Pixel 3XL currently on May update and Magisk 19.2 Canary (19203).

When I install this app on the latest beta it tells me that the daemon is not running although it is. Did I miss something in the install?

breucode, 2019-06-07:
I have the same Problem on Magisk 19.3 on the latest LineageOS for beryllium.

MatteCarra, 2019-06-09:
The procedure should be ok.

I'm preparing for 3 big exams and I plan to continue developing the app once they are all finished.

You could try to install acc from the magisk repo

VR-25, 2019-06-09:
@bridges86406, @breucode,
- Did you install `ACCApp 1.0.8` and `ACC 2019.6.8`? These are the latest versions (time of this writing).
- Has ACCApp been granted root access?
- Did you try clearing the app's data?

MatteCarra, 2019-07-11:
I'm closing this until we have more details.
All compatibility issues should be solved with next builds.

closed 2019-07-11 by MatteCarra