MatteCarra/AccA #27: Negative values when charging battery

Nachtfalkeaw, 2019-06-07:
With 1.0.8 and miui 10, Android 9, Xiaomi Mi8 and latest Magister (19.3/19.2) i can See negative values when charging the battery. It is charging -2000 instead +2000

MatteCarra, 2019-06-09:
Thanks a lot for the report.

I'm preparing for 3 big exams.
I plan to continue developing the app once they are all finished.

VR-25, 2019-06-09:
@Nachtfalkeaw, what's the output of the command `su -c acc --info? while plugged in?

Nachtfalkeaw, 2019-06-12:
NAME=battery
INPUT_SUSPEND=0
STATUS=Charging
HEALTH=Good
PRESENT=1
CHARGE_TYPE=Fast
CAPACITY=50
CHARGER_TEMP=325
CHARGER_TEMP_MAX=803
INPUT_CURRENT_LIMITED=1
VOLTAGE_NOW=3888173
VOLTAGE_MAX=4400000
VOLTAGE_QNOVO=-22
CURRENT_NOW=-326171
CURRENT_QNOVO=-22
CONSTANT_CHARGE_CURRENT_MAX=3300000
TEMP=290
TECHNOLOGY=Li-poly
STEP_CHARGING_ENABLED=0
SW_JEITA_ENABLED=1
CHARGE_DONE=0
PARALLEL_DISABLE=0
SET_SHIP_MODE=0
DIE_HEALTH=Cool
RERUN_AICL=0
CHARGER_TYPE=9
DP_DM=0
CHARGE_CONTROL_LIMIT_MAX=16
CHARGE_CONTROL_LIMIT=0
CHARGE_COUNTER=1528169
DC_THERMAL_LEVELS=0
CHARGE_FULL=3346000
CYCLE_COUNT=176

VR-25, 2019-06-24:
`CURRENT_NOW=-326171`: this suggests that your system is the source of the issue.
You may want to check that line at different times and report your findings.

VR-25, 2019-08-14:
@Nachtfalkeaw, any update on this, in regards to my previous comment?

closed 2019-09-28 by MatteCarra

bollocks-fks, 2020-04-14:
This is how physicists and electrical engineers mathematically describe which way the electrons are flowing. It's similar to velocity. Reporting negative current while charging is the behavior of the Redmi Note 6 Pro as well. This is likely the driver writer's intended behavior. This is how physicists and electrical engineers mathematically describe which way the electrons are flowing.

On a related note, I just installed the latest version of ACC from the git today. I notice that it's calculating power now. However the sign in the output of doesn't change when switching from charging to discharging. That sign should also change depending which way energy is transferring. Just multiply the numbers like normal and you get the right result

goduck777, 2021-03-18:
I have the same issue on Oneplus 8. The charging current is always negative. The charging voltage is positive.

I am using the stock ROM. Do you think this is a kernel issue?