MatteCarra/AccA #35: Wrong current when running on usb power

kmarius, 2019-06-22:
I am using the `battery/charge_disable` switch. When the battery is not being charged and the phone is running on usb power the app will show the wrong current: the system will draw about 2mA but the app shows -2000mA instead.

The responsible function is this one:
https://github.com/MatteCarra/AccA/blob/b14a4d48456b684b1746a81d2b8689cb70d34cff/app/src/main/java/mattecarra/accapp/models/BatteryInfo.kt#L92

I'm running Pie on a Pixel 3a and since the current in the in the corresponding sysfs file is given un uA, it should always be divided by 1000. I don't know if there are other phones/version where the current is given in mA already.

VR-25, 2019-06-24:
Most systems use uA.
Could you show us the output of `acc -i` in that situation?
Thanks

kmarius, 2019-06-24:
Output off `acc -i` while running on usb power:
```
NAME=battery
INPUT_SUSPEND=0
STATUS=Not charging
HEALTH=Good
PRESENT=1
CHARGE_TYPE=N/A
CAPACITY=44
CHARGER_TEMP=364
CHARGER_TEMP_MAX=812
INPUT_CURRENT_LIMITED=0
VOLTAGE_NOW=3815175
VOLTAGE_MAX=4200000
VOLTAGE_QNOVO=-22
CURRENT_NOW=976
CURRENT_QNOVO=-22
CONSTANT_CHARGE_CURRENT_MAX=2950000
TEMP=313
TECHNOLOGY=Li-ion
STEP_CHARGING_ENABLED=0
SW_JEITA_ENABLED=0
TAPER_CONTROL_ENABLED=0
CHARGE_DISABLE=1
CHARGE_DONE=0
PARALLEL_DISABLE=0
SET_SHIP_MODE=0
DIE_HEALTH=Cool
RERUN_AICL=0
DP_DM=0
CHARGE_CONTROL_LIMIT_MAX=6
CHARGE_CONTROL_LIMIT=0
CHARGE_COUNTER=1305874
INPUT_CURRENT_MAX=1500000
CHARGE_FULL=3058000
CYCLE_COUNT=13
```

MatteCarra, 2019-06-25:
In the future I will adopt a different logic: I will take a bunch of CURRENT_NOW values in 10 seconds and I will do the avarage and use this logic: abs(current_avg) > 10000 => current is expressed in uAh
If it's expressed in uAh i will write it in the app database and ALWAYS divide by /1000

Current solution is not good, because it can oscillate over time

MatteCarra, 2019-08-09:
Please let me know if you still have any issue when a new release is available.

closed 2019-08-09 by MatteCarra