MatteCarra/AccA #36: Can't read it's own config files apparently

ovizii, 2019-06-24:
Can't read it's own config files apparently. Tried version 1.0.6 and 1.0.8

[image]

squabbi, 2019-06-24:
Try running `acc -s r` to restore the default configuration.

VR-25, 2019-06-24:
The latest acc uses a new config filename (acc.conf). That's the problem.
I totally forgot the app! :-o

ovizii, 2019-06-24:
sorry guys I've never touched the command line of this magisk module due to using the app. 
I guess I'll just wait for a newer version of the app then :-) 

Feel free to close this issue once the app knows about the new file name.

VR-25, 2019-06-24:
Meanwhile, you can downgrade acc to [201906200](https://github.com/VR-25/acc/releases/download/201906200/acc-201906200.zip). If you're doing that, run `acc -s wakeUnlock` afterwards.

MatteCarra, 2019-06-24:
Fixed: https://github.com/MatteCarra/AccA/releases/tag/v1.0.9

closed 2019-06-24 by MatteCarra

ovizii, 2019-06-24:
Thanks, works perfectly.