MatteCarra/AccA #39: Feature suggestions

siffreinsg, 2019-06-27:
Hey!
I really enjoy the app but right now it does not fit exactly my usage of ACC and I'd like to suggest some new features :)

The biggest one is the integration with Tasker to be able to switch between profiles without human interaction. For instance, at night I want my phone to charge very slowly in 3 to 5 hours but the day I want my phone to charge faster. If making AccA a Tasker plugin is a bit hard, another solution might just be to store profiles created in the app in a folder in /sdcard/acc and then using Tasker to swap profiles using "Copy file" action.

My second suggestion is a persistant notification showing charging state, battery current, active profile (using name), and even more information when expanding the notification such as cooldown state, battery temp, ...

I would also like to suggest a "dev mode" which notify the user if acc daemon crashes in background with the error message if applicable. This could be very useful to detect flaws in profiles or why the daemon is not working sometime :)

And finally a more user-friendly "log" section in the app to understand what ACC is doing (for example : Battery temp = 40°C => Pause charging for 30s \nResuming charging.\nBattery level > 80%, enabling cooldown\n........) x) (I think I'm not the only who don't understand anything when looking in this section and would like something more explicit ^^)

Thank you very much for all the work you are doing, you are awesome <3
Thanks to VR25 too for the work done on the daemon <3

squabbi, 2019-06-28:
Yes the logging is something that I've had in mind as well! Also Tasker integration would be cool.

kmarius, 2019-07-01:
The quickest way to achieve basic Tasker integration would be allowing to change profiles with broadcast intents. I hacked something together here https://github.com/kmarius/AccA/commit/7fff088e049651a3ebb7c66151cb86299e3e7e13 (couldn't figure out how also apply active profile in the Acca settings), so this can be done with little effort.

siffreinsg, 2019-07-01:
Owh that's a good idea ! I'll try it asap :)
Thanks

undercoverturbo, 2019-07-03:
You can use a Macro APP like MacroDroid. As long As it can execute shell commands/scripts.
Then you can just put in "acc 46 44 && acc --set applyOnPlug usb/current_max:500000"
or something similar like that.
When you trigger it (e.g. on a specific time of day) before ging to bed and then trigger something like "acc 80 70 && acc --set applyOnPlug usb/current_max:2200000" in the morning again, you have the desired functionality.

siffreinsg, 2019-07-03:
This is what I'm doing actually... But I would like to be able to change profiles from the AccA app instead of editing the task and commands.

squabbi, 2019-07-04:
@siffreinsg With DJS (which will be implemented into the app later on) you should be able to schedule ACC actions. But if I'm not mistaken it may only have time based events.

@kmarius that's great! I'll keep that in mind when we get over to taking a proper look at Tasker integration. 👍

cxyzzz, 2019-07-22:
> 这就是我实际做的......但我希望能够从AccA应用程序更改配置文件，而不是编辑任务和命令。

the busybox have `crond` and `crontab`,it's can do it

MatteCarra, 2019-08-22:
@siffreinsg Do you think that the new improved version of schedules is enough to avoid using Tasker or do you there's still the need of using it?

siffreinsg, 2019-08-23:
I'm using tasker to change the way the phone charge if i'm home or not, if it's day or night, if i have an agenda event soon, ... so it's more than just time-based profiles.
The app would have allowed me to modify the profiles more easily and on the fly but it's more a matter of comfort than feature as I'm currently using commands to do the same ^^
Thanks anyway :)

lukegluke, 2020-12-20:
While we all here waiting for a full AccA Tasker plugin, another solution is to SQL query acca_database to find profile by name, parse settings and then run a acc script command. Something like this (it is just first step not complete but working preparation):
```
    AccA set profile (16)
    	A1: Copy Dir [ From://data/data/mattecarra.accapp/databases To:Tasker/cache/tmp/accapp Use Root:On ] 
    	A2: SQL Query [ Mode:Formatted File:Tasker/cache/tmp/accapp/acca_database Table:profiles_table Columns:configCapacity, configTemperature Query:profileName = '%PROFILE' Selection Parameters: Order By: Output Column Divider: Variable Array:%dbreturn Use Root:Off ] 
    	A3: Flash [ Text:"%PROFILE" profile not found! Long:Off ] If [ %dbreturn(#) < 1 ]
    	A4: Return [ Value:-22 Stop:On ] If [ %dbreturn(#) < 1 ]
    	A5: JavaScriptlet [ Code://parse result %dbreturn(1) here
    
    alert(dbreturn[0])
    
    //form the script line here Libraries: Auto Exit:On Timeout (Seconds):10 ] 
    	A6: Run Shell [ Command:/dev/.vr25/acc/acca --set pause_capacity=20 resume_capacity=15 Timeout (Seconds):0 Use Root:On Store Output In:%out Store Errors In:%errout Store Result In: Continue Task After Error:On ] 
    	A7: Flash [ Text:%errout Long:On ] If [ %errout Set ]
    	A8: Return [ Value:-5 Stop:On ] If [ %errout Set ]
```
(_coping dir in A1 just a workaround. For some reason when Tasker copy database file directly or I did through root explorer manually it was empty database (why?)_)

In fact AccA app in this case become just an ui for profile creation and then Tasker duplicate the same backend job.
Also Tasker can have gui, so if one has time, an alternative to AccA  app could be created in Tasker itself :)