MatteCarra/AccA #40: Added German in crowdin

undercoverturbo, 2019-07-03:
I added a German translation to the crowdin project.
How do I get it to git/how do you do that?
Not used to work with crowdin :)

MatteCarra, 2019-07-04:
Thanks!
I see all of the translations, but there a some minor errors.
I think I've to approve the changes and then they will be automatically pushed here

undercoverturbo, 2019-07-04:
What errors do you mean?
If you tell me, I can fix them.

MatteCarra, 2019-07-04:
I meant that Crowdin has some errors.
I don't know German and I wouldn't be able to correct you.
Example: Akku-Statistiken beim Ausstecken zurücksetzen = Spellcheck failed for the following word: Ausstecken
https://crowdin.com/proofread/advanced-charging-controller/all/en-de

undercoverturbo, 2019-07-04:
I've checked the errors Crowdin ist showing there.
They all seem to be fine. I mean it also fails to spellcheck words like App.

MatteCarra, 2019-07-04:
Ok thanks for the confirmation.
I will merge them as soon as understand how to use crowdin.
We haven't made it public yet because the app will be undergoing major changes in August

closed 2019-07-04 by MatteCarra

MatteCarra, 2019-07-11:
Done!
Next release will have your strings.