MatteCarra/AccA #46: Implement Scheduling in to ACCA

squabbi, 2019-07-11:
Use DJS to integrate scheduling of commands in ACCA.

VR-25, 2019-07-12:
All of my projects are being built upon the exact same framework and base API - which means they're going to have the same core features found on ACC.
I'm debating... should DJS be bundled into ACC itself, or AccA?

cxyzzz, 2019-07-22:
> 我的所有项目都是基于完全相同的框架和基础API构建的 - 这意味着它们将具有与ACC相同的核心功能。
> 我正在辩论...... DJS应该捆绑到ACC本身，还是AccA？

This can use busybox,it's have `crond` and `crontab`

VR-25, 2019-08-14:
> > 我的所有项目都是基于完全相同的框架和基础API构建的 - 这意味着它们将具有与ACC相同的核心功能。
> > 我正在辩论...... DJS应该捆绑到ACC本身，还是AccA？
> 
> This can use busybox,it's have `crond` and `crontab`
`cron` doesn't look very friendly to me at this point.

closed 2019-08-16 by MatteCarra