MatteCarra/AccA #48: Installation failed

rodylg, 2019-07-19:
SU permission has been granted at prompt and yet this warning is displayed despite having an active Internet connection (WiFi/data) and even after installing acc through the magisk manager (7.3.2). The magisk (19.3) module is working and up to date (201907180).

===System's data===
Model: SM-J700M (j7elte)
TWRP: 3.2.3-0
Magisk: 19.3
Device API level: 23
Android version: 6.0.1
Kernel Version: 3.10.61-11132510
Build number: MMB29K.J700MUBS2BQF4
SELinux status: Enforcing (SEPF_SECMOBILE_6.0.1_0034)
ASKS v1.2_161011
KNOX 2.6

MatteCarra, 2019-07-19:
Please upload this file /data/local/tmp/acc-install.log after the installation fails.

Everyone with the same issue should do the same.

Thanks

MatteCarra, 2019-07-19:
Btw acc module magisk won't be required to be installed separately anymore and is included by vr25 in the apk.

For more details read the changelog

rodylg, 2019-07-20:
Contents of acc-install.log

+ >/dev/null 
+ which busybox
+ id -u
+ '[' 0 -ne 0 ']'
+ id=acc 
+ set -e
+ '[' -f //install-tarball.sh ']'
+ cd /data/user/0/mattecarra.accapp/files
+ '[' -d acc/acc-init.sh ']'
+ rm -rf acc-201907180/
+ tar -xf acc_bundle.tar.gz
+ export 'installDir0='
+ sh acc-201907180/install-current.sh

name=Advanced Charging Controller (acc) version=2019.7.18
Copyright (c) 2017-2019, author=VR25 (xda-developers.com)
License: GPLv3+

(i) Installing to /data/data/mattecarra.accapp/files/id=acc/...
cp: acc-201907180/id=acc/: No such file or directory
cp: bad 'acc-201907180/id=acc/': No such file or directory

VR-25, 2019-07-20:
This seems to be a busybox related issue.
More specifically, the installer may be using the system's `sed` (often problematic) over busybox's.
The mechanism is being improved.
Could you contact me on telegram (`vr25xda`) to receive debug AccA builds - so we can work through this faster?
Thanks

closed 2019-07-21 by VR-25

stonerl, 2019-09-15:
Having the same issue on my 1+3. Here is the log

```
+ '[' -d /sbin/.magisk/busybox ']'
+ '[' -d /sbin/.core/busybox ']'
+ >/dev/null 
+ which busybox
+ mkdir -p -m 700 /dev/.busybox
+ busybox install -s /dev/.busybox
BusyBox v1.30.1-meefik (2019-03-03 03:38:18 UTC) multi-call binary.

Usage: install [-cdDsp] [-o USER] [-g GRP] [-m MODE] [-t DIR] [SOURCE]... DEST

Copy files and set attributes

	-c	Just copy (default)
	-d	Create directories
	-D	Create leading target directories
	-s	Strip symbol table
	-p	Preserve date
	-o USER	Set ownership
	-g GRP	Set group ownership
	-m MODE	Set permissions
	-t DIR	Install to DIR
+ PATH=/dev/.busybox:/sbin:/system/sbin:/system/bin:/system/xbin:/odm/bin:/vendor/bin:/vendor/xbin 
+ id -u
+ '[' 0 -ne 0 ']'
+ id=acc 
+ set -e
+ '[' -f //install-tarball.sh ']'
+ cd /data/user/0/mattecarra.accapp/files
+ '[' -d acc/acc-init.sh ']'
+ rm -rf acc-201907211/
+ tar -xf acc_bundle.tar.gz
+ export 'installDir0='
+ sh acc-201907211/install-current.sh

Advanced Charging Controller (acc) 2019.7.21-r1
Copyright (c) 2017-2019, VR25 (xda-developers.com)
License: GPLv3+

(i) Installing to /data/data/mattecarra.accapp/files/acc/...
cp: /data/data/mattecarra.accapp/files/acc/: Is a directory
```