MatteCarra/AccA #50: files under /sdcard/

0xRR, 2019-07-20:
There are more and more files and folder under /sdcard/ .
It will be very thankful if you put them under /sdcard/Android/data/mattecarra.accapp .

VR-25, 2019-07-21:
Which folders are you referring to, specifically?
The app doesn't store anything in user storage.
In fact, it doesn't even have user storage access permissions.

0xRR, 2019-07-21:
ACCA_v1.0.11(16) has only an folder names "/acc" .
ACCA_v1.0.12(17) has an folder names "/acc-data" and a file names “.acc-config-backup.txt” .

VR-25, 2019-07-21:
These are created by the back-end (acc). You can safely delete `/sdcard/acc-data/`. It's an artifact.

Now we're left with `/sdcard/.acc-config-backup.txt`. This file is there to survive TWRP factory resets. The actual config is stored at `/data/adb/acc-data/config.txt`.

The backup file can't be stored at `/sdcard/Android/data/mattecarra.accapp/`, because ACC itself is independent. Besides, this would interfere with other front-ends.

0xRR, 2019-07-21:
> These are created by the back-end (acc). You can safely delete `/sdcard/acc-data/`. It's an artifact.
>
> Now we're left with `/sdcard/.acc-config-backup.txt`. This file is there to survive TWRP [...]

Thanks for the explanation ! 

I see , my fault , apologises !

closed 2019-07-21 by VR-25