MatteCarra/AccA #52: Charge till 90% keeps charging bug

dkoek, 2019-07-21:
Since acc doesn't install by Magisk but directly from acc apk 1.0.13 The app doesn't stop or limit charging. Deinstall and reinstall doesn't help the problem. 
[image]

MatteCarra, 2019-07-21:
Try to delete app data and to install latest version: https://github.com/MatteCarra/AccA/releases/tag/v1.0.14

VR-25, 2019-07-22:
@dkoek, what's the output of `su -c acc -s` after applying that profile?

dkoek, 2019-07-22:
Deleted data and cache. 
Sorry don't know where to install the zip file of the 1.0.14 version . Till now I used the apk file.

closed 2019-07-22 by VR-25