MatteCarra/AccA #62: OxygenOS Car Charge Notification doesn't recognize when ACC shuts off charging

moonspeak, 2019-08-15:
OxygenOS on the OnePlus 7 Pro has a persistent notification that appears when you plug in the official _Warp Charge 30 car charger_. It displays this notification until the car charger is disconnected.

Unfortunately, it seems that however ACC/AccA handles the 'stop' functionality goes undetected by the process/logic in charge of removing the notification, so when ACC disables charging, the persistent notification remains. Then even when physically removing the cable, the phone still does not realize and the notification is stuck until I either charge with a wall charger or do use the car charger and remember to unplug it before ACC disables charging.

I would simply disable this notification, but the system does not allow it to be disabled. I could also use Automagic to consistently snooze it to suppress it, but I'm also concerned that it could potentially cause some other issue such as a background process that may still think the phone is being charged which may lead to unintended consequences.

I'm hoping there's some way to understand and fix this, though I'm not sure how and I know it would be impossible to test if you guys don't have this phone and the car charger, but I'm fully willing to coordinate, provide logs, and test any ideas or test versions you come up with.

VR-25, 2019-08-16:
I would like you to perform a few tests.

1. Does the persistent notification disappear (while the charger is _unplugged_) after running `acc -e`?

2. Does enforcing a particular charging switch help?
- This can be done with `acc -s s` or from the app.
- Test each switch with `acc -d` and `acc -e` (disable and enable charging).

moonspeak, 2019-08-21:
> I would like you to perform a few tests.
>
> 1. Does the persistent notification disappear (while the charger is _unplugged_) after running `acc -e`?
> 2. Does enforcing a particular charging switch [...]

1. No, unfortunately it remains
2. None of the three switches available by default seem to make a difference. I only have three by default in the dialog in the app.
- battery/input_suspend 0 1 - Doesn't work
- battery/chargin_enabled 1 0 - Makes no difference
- dc/input_suspend 0 1 - Makes no difference

I know there are [a lot of other switches](https://github.com/VR-25/acc/blob/master/acc/switches.txt) available, but I'm not sure where to begin testing any of them or if trying any out could cause damage to the phone/battery in some way. Is there any more info on those anywhere?

VR-25, 2019-12-14:
Somewhere under Android settings`, there's [Developer options](https://developer.android.com/studio/debug/dev-options) > Running services.
@moonspeak, could you identify the package name associated with the service responsible for that notification?

closed 2020-03-09 by VR-25