MatteCarra/AccA #63: Voltage Control File Select

droid-jd, 2019-08-22:
When I attempt to select a Voltage control file, I am unable to see the actual file being selected, or the options available to me. This happened in both V. 14 and V. 15 of the app. I believe it is a design issue only, and may not require the submission of a log, but I have attached screenshots to explain what I'm experiencing.
On the Portrait screenshot, I am unable to scroll through the other options for a Voltage Control file.
[image]

On the Landscape screenshot, I can see that there are five (5) options for voltage control files, but I cannot actually see what they are.
[image]

I have not seen anyone else report this issue in the XDA thread, and the quick browsing of "related issues" on github did not produce any immediately obvious results. 
I am running LineageOS 16 on my OnePlus 6T, in case that information is helpful. 

By contrast, the Capacity Control/Charging Switch menu pops up perfectly for me, with clear listings of the available switch files. 
[image]

If there is anything else you need from me, please do not hesitate to let me know.

MatteCarra, 2019-08-22:
Thanks a lot for the very detailed bug report.
I've never seen this issue and I can't reproduce it on my device (OnePlus 6, OxygenOS).
My hypothesis is that it's a strange LOS bug, but I will check if I can do anything about it.

[image]
[image]

droid-jd, 2019-08-22:
Happy to help! Been using the app and module for some time now, so it's
nice to be able to contribute something back.

By the way, seeing as we have similar devices, are you able to get battery
idle mode or voltage limiting working?
Apologies if this isn't the appropriate venue to discuss this.

MatteCarra, 2019-08-22:
None of them sadly.
These features depend on the kernel and oneplus kernel does not seem to support them.

closed 2020-03-09 by VR-25