MatteCarra/AccA #65: Phone idle getting disturbed

knreddypc, 2019-08-31:
Before installation of this app, (acc is installed and effective) Automat App is able to shutdown my rooted tab if the tab is idle or inactive for more than 3 hours 

Auto idle shutdown is not working after installation of AccA I.e. it is disturbing the sleep or idle period of tab

MatteCarra, 2019-08-31:
Do you have the same problem if you install Acc without AccA?

knreddypc, 2019-08-31:
Dear Sir, you have taken great project like GUI for acc. It is highly
appreciable.
I was using Advanced_Charging_Controller_(acc)-2018.12.10.zip
Automate was working fine with this acc.

Flawless but latest versions were not working well like it will not at high
limit of charge etc.

Sir now I will uninstalled AccA and recheck all possibilities.

Regards
K. N. Reddy

MatteCarra, 2019-08-31:
Thanks a lot!
I asked this question because we need to know if the problem is acc or AccA.
Once we have isolated the problem we can try to fix it.

knreddypc, 2019-09-03:
Dear Sir,
I have rechecked the issue by reinstalling separately latest acc file and
 AccA with default acc, they are working normal.

The issue may relavent to installation of AccA over old( 2018 )  acc  file
(as installation locations are changed)

Regards

MatteCarra, 2019-09-03:
Ok, great!
Thanks for the update

closed 2019-09-03 by MatteCarra