MatteCarra/AccA #69: Invisible/Overlayed overview button

okmiim, 2019-09-17:
The android overview buttons are not visible, however you can still use them (by guessing/knowing their positions).
It seems like that the buttons are overlayed by the white ui, however the tabs are positioned above the button area.

ACCA Version: 1.0.16 (21)
ACC Version: 201908180
Android Version: 6.0

squabbi, 2019-09-18:
Would I grab a screenshot from you? I have seen it before but it seems to be even worse by the way you're describing it.

I do have some fixes in mind right now, but won't be able to test them until later.

zputnyq, 2019-09-18:
This happen to my device as well, especially for white theme. So far, it's ok black them
I'm on android 6.0.1 sony xperia m4.

okmiim, 2019-09-18:
I added some screenshots. 
As @zputnyq mentioned, the dark theme works fine.
Note: The image compression doesn't change the visibility on the light theme.

[image][image][image]