MatteCarra/AccA #71: Dark theme does not show selected profile

bhafer, 2019-10-20:
When using dark theme and selective a profile, the highlighted line does not appear to indicate the selected profile.

Steps to reproduce:
1) Open settings and select the dark theme.
2) Go to the Profiles screen.
3) Tap to select a new profile.

Expected Outcome:
-- A green line should appear to indicate the tapped profile is selected.

Actual Outcome:
-- No green line appears.

MatteCarra, 2019-10-20:
Thanks for the report.
I think that he problem is not the theme, but one of the latest commit to detect if profile was changed (for example by a scheduled).
I will inspect it

bhafer, 2019-10-20:
Thank you.
Except it is working correctly for me with the light theme.

bhafer, 2019-10-21:
Update: Hmm, yeah. Maybe you're right. It appears that the issue is intermittent or that sometimes there is just a long day before the green line appears, with either theme.

MatteCarra, 2019-10-21:
Let me know!

closed 2019-10-21 by MatteCarra