MatteCarra/AccA #72: Acc daemon stops when the app is closed

paypur, 2019-10-22:
upgrading to 1.0.19 has caused this issue and downgrading to 1.0.14 fixes it

MatteCarra, 2019-10-22:
I can't reproduce your bug.
How do you check if daemon is running when the app is in background?
You should use 'accc -D' from terminal.

VR-25, 2019-10-22:
@paypur, also, please send us `/sdcard/acc-logs-*.tar.bz2` after running `acc -l -e`.

paypur, 2019-10-22:
> @paypur, also, please send us `/sdcard/acc-logs-*.tar.bz2` after running `acc -l -e`.

[logs.zip](https://github.com/MatteCarra/AccA/files/3758853/logs.zip)

deulizealand, 2019-10-29:
I have face that to , but when i test it. (start and stop limit) it work as well . Also when you stop the service and reopen the app.  The service doesnt start. It's differrnt when you start the service and reopen the app.  Service is start but there is some delay to check it.  Hope you understand what i mean .

paypur, 2019-11-10:
After further testing acc does work, but is goes over my set limit by a couple of percent.

closed 2019-12-14 by VR-25