MatteCarra/AccA #73: Installation failed

KreativeKrise, 2019-10-27:
After I installed the latest version 10.0.20 the installation of the acc daemon fails.

```
+ trap 'e=$?; echo; exit $e' EXIT
+ '[' -d /sbin/.magisk/busybox ]
+ '[[' /sbin/.magisk/busybox:/sbin:/system/sbin:/product/bin:/apex/com.android.runtime/bin:/system/bin:/system/xbin:/odm/bin:/vendor/bin:/vendor/xbin '==' /sbin/.magisk/busybox ]]
+ PATH=/sbin/.magisk/busybox:/sbin/.magisk/busybox:/sbin:/system/sbin:/product/bin:/apex/com.android.runtime/bin:/system/bin:/system/xbin:/odm/bin:/vendor/bin:/vendor/xbin
+ id -u
+ '[' 0 -ne 0 ]
+ set -euo pipefail
+ '[' -f /data/data/mattecarra.accapp/files/install-current.sh ]
+ srcDir=acc-201910132
+ '[' '!' -f acc-201910132/module.prop ]
+ print name
+ sed -n 's|^name=||p' acc-201910132/module.prop
+ name='Advanced Charging Controller (acc)'
+ print author
+ sed -n 's|^author=||p' acc-201910132/module.prop
+ author='VR25 (xda-developers.com)'
+ print version
+ sed -n 's|^version=||p' acc-201910132/module.prop
+ version=2019.10.13-r2-dev
+ print versionCode
+ sed -n 's|^versionCode=||p' acc-201910132/module.prop
+ versionCode=201910132
+ installDir=/data/data/mattecarra.accapp/files
+ config=/data/media/0/acc/acc.conf
+ '[' -f /data/media/0/acc/acc.conf ]
+ mv /data/media/0/acc/config.txt /data/media/0/acc/acc.conf
+ :
+ '[' -d /data/media/0/acc ]
+ config=/data/adb/acc-data/config.txt
+ '[' -f /data/adb/acc-data/config.txt ]
+ cp /data/media/0/.acc-config-backup.txt /data/adb/acc-data/config.txt
+ print versionCode /data/adb/acc-data/config.txt
+ configVer=201907090
+ '[' -d /data/data/mattecarra.accapp/files ]
+ '[' -d /data/data/mattecarra.accapp/files ]
+ '[' -d /data/data/mattecarra.accapp/files ]
+ '[' -d /data/data/mattecarra.accapp/files ]
+ cat
+ pkill -9 -f '/acc (-|--)|/accd.sh'
+ :
+ readlink -f /sbin/.acc/acc
+ rm -rf /data/data/mattecarra.accapp/files/acc
+ cp -R acc-201910132/acc/ /data/data/mattecarra.accapp/files/
+ installDir=/data/data/mattecarra.accapp/files/acc
+ '[' x '==' x ]
+ installDir0=/data/data/mattecarra.accapp/files/acc
+ cp acc-201910132/module.prop /data/data/mattecarra.accapp/files/acc/
+ mkdir -p /data/adb/acc-data/info
+ cp -f acc-201910132/License.md acc-201910132/README.md /data/adb/acc-data/info
+ mv /data/data/mattecarra.accapp/files/acc/service.sh /data/data/mattecarra.accapp/files/acc/acc-init.sh
+ '[' -f /data/adb/acc-data/config.txt ]
+ '[' 201907090 -lt 201910130 ]
+ rm /data/adb/acc-data/config.txt
+ cp -f acc-201910132/bin/acc-uninstaller.zip /data/media/0/
+ set_perms_recursive /data/adb/acc-data
+ local 'owner=0' 'target='
+ find /data/adb/acc-data
+ read target
+ set_perms /data/adb/acc-data 0
+ readlink -f /data/adb/acc-data
+ local 'owner=0' 'perms=0600' 'target=/data/adb/acc-data'
+ echo /data/adb/acc-data
+ grep -q '.*\.sh$'
+ '[' -d /data/adb/acc-data ]
+ perms=0700
+ chmod 0700 /data/adb/acc-data
+ chown 0:0 /data/adb/acc-data
+ restorecon /data/adb/acc-data
+ read target
+ set_perms /data/adb/acc-data/info 0
+ readlink -f /data/adb/acc-data/info
+ local 'owner=0' 'perms=0600' 'target=/data/adb/acc-data/info'
+ echo /data/adb/acc-data/info
+ grep -q '.*\.sh$'
+ '[' -d /data/adb/acc-data/info ]
+ perms=0700
+ chmod 0700 /data/adb/acc-data/info
+ chown 0:0 /data/adb/acc-data/info
+ restorecon /data/adb/acc-data/info
+ read target
+ set_perms /data/adb/acc-data/info/License.md 0
+ readlink -f /data/adb/acc-data/info/License.md
+ local 'owner=0' 'perms=0600' 'target=/data/adb/acc-data/info/License.md'
+ echo /data/adb/acc-data/info/License.md
+ grep -q '.*\.sh$'
+ '[' -d /data/adb/acc-data/info/License.md ]
+ chmod 0600 /data/adb/acc-data/info/License.md
+ chown 0:0 /data/adb/acc-data/info/License.md
+ restorecon /data/adb/acc-data/info/License.md
+ read target
+ set_perms /data/adb/acc-data/info/README.md 0
+ readlink -f /data/adb/acc-data/info/README.md
+ local 'owner=0' 'perms=0600' 'target=/data/adb/acc-data/info/README.md'
+ echo /data/adb/acc-data/info/README.md
+ grep -q '.*\.sh$'
+ '[' -d /data/adb/acc-data/info/README.md ]
+ chmod 0600 /data/adb/acc-data/info/README.md
+ chown 0:0 /data/adb/acc-data/info/README.md
+ restorecon /data/adb/acc-data/info/README.md
+ read target
+ set_perms /data/adb/acc-data/logs 0
+ readlink -f /data/adb/acc-data/logs
+ local 'owner=0' 'perms=0600' 'target=/data/adb/acc-data/logs'
+ echo /data/adb/acc-data/logs
+ grep -q '.*\.sh$'
+ '[' -d /data/adb/acc-data/logs ]
+ perms=0700
+ chmod 0700 /data/adb/acc-data/logs
+ chown 0:0 /data/adb/acc-data/logs
+ restorecon /data/adb/acc-data/logs
+ read target
+ set_perms /data/adb/acc-data/logs/install-tarball.sh.log 0
+ readlink -f /data/adb/acc-data/logs/install-tarball.sh.log
+ local 'owner=0' 'perms=0600' 'target=/data/adb/acc-data/logs/install-tarball.sh.log'
+ echo /data/adb/acc-data/logs/install-tarball.sh.log
+ grep -q '.*\.sh$'
+ '[' -d /data/adb/acc-data/logs/install-tarball.sh.log ]
+ chmod 0600 /data/adb/acc-data/logs/install-tarball.sh.log
+ chown 0:0 /data/adb/acc-data/logs/install-tarball.sh.log
+ restorecon /data/adb/acc-data/logs/install-tarball.sh.log
+ read target
+ set_perms /data/adb/acc-data/logs/install.log 0
+ readlink -f /data/adb/acc-data/logs/install.log
+ local 'owner=0' 'perms=0600' 'target=/data/adb/acc-data/logs/install.log'
+ echo /data/adb/acc-data/logs/install.log
+ grep -q '.*\.sh$'
+ '[' -d /data/adb/acc-data/logs/install.log ]
+ chmod 0600 /data/adb/acc-data/logs/install.log
+ chown 0:0 /data/adb/acc-data/logs/install.log
+ restorecon /data/adb/acc-data/logs/install.log
+ read target
+ set_perms /data/adb/acc-data/logs/init.log 0
+ readlink -f /data/adb/acc-data/logs/init.log
+ local 'owner=0' 'perms=0600' 'target=/data/adb/acc-data/logs/init.log'
+ echo /data/adb/acc-data/logs/init.log
+ grep -q '.*\.sh$'
+ '[' -d /data/adb/acc-data/logs/init.log ]
+ chmod 0600 /data/adb/acc-data/logs/init.log
+ chown 0:0 /data/adb/acc-data/logs/init.log
+ restorecon /data/adb/acc-data/logs/init.log
+ read target
+ chmod 0666 /data/media/0/acc-uninstaller.zip
+ pkg=/data/data/mattecarra.accapp
+ pkg=mattecarra.accapp
+ grep mattecarra.accapp /data/system/packages.list
+ awk '{print $2}'
+ owner=10004
+ set_perms_recursive /data/data/mattecarra.accapp/files 10004
+ local 'owner=10004' 'target='
+ read target
+ find /data/data/mattecarra.accapp/files
+ set_perms /data/data/mattecarra.accapp/files 10004
+ readlink -f /data/data/mattecarra.accapp/files
+ local 'owner=10004' 'perms=0600' 'target=/data/data/mattecarra.accapp/files'
+ echo /data/data/mattecarra.accapp/files
+ grep -q '.*\.sh$'
+ '[' -d /data/data/mattecarra.accapp/files ]
+ perms=0700
+ chmod 0700 /data/data/mattecarra.accapp/files
+ chown 10004:10004 /data/data/mattecarra.accapp/files
+ restorecon /data/data/mattecarra.accapp/files
+ read target
+ set_perms /data/data/mattecarra.accapp/files/profiles 10004
+ readlink -f /data/data/mattecarra.accapp/files/profiles
+ local 'owner=10004' 'perms=0600' 'target=/data/data/mattecarra.accapp/files/profiles'
+ echo /data/data/mattecarra.accapp/files/profiles
+ grep -q '.*\.sh$'
+ '[' -d /data/data/mattecarra.accapp/files/profiles ]
+ chmod 0600 /data/data/mattecarra.accapp/files/profiles
+ chown 10004:10004 /data/data/mattecarra.accapp/files/profiles
+ restorecon /data/data/mattecarra.accapp/files/profiles
+ read target
+ set_perms /data/data/mattecarra.accapp/files/Standard.profile 10004
+ readlink -f /data/data/mattecarra.accapp/files/Standard.profile
+ local 'owner=10004' 'perms=0600' 'target=/data/data/mattecarra.accapp/files/Standard.profile'
+ echo /data/data/mattecarra.accapp/files/Standard.profile
+ grep -q '.*\.sh$'
+ '[' -d /data/data/mattecarra.accapp/files/Standard.profile ]
+ chmod 0600 /data/data/mattecarra.accapp/files/Standard.profile
+ chown 10004:10004 /data/data/mattecarra.accapp/files/Standard.profile
+ restorecon /data/data/mattecarra.accapp/files/Standard.profile
+ read target
+ set_perms /data/data/mattecarra.accapp/files/Voll aufladen.profile 10004
+ readlink -f /data/data/mattecarra.accapp/files/Voll
+ local 'owner=aufladen.profile' 'perms=0600' 'target=/data/data/mattecarra.accapp/files/Voll'
+ echo /data/data/mattecarra.accapp/files/Voll
+ grep -q '.*\.sh$'
+ '[' -d /data/data/mattecarra.accapp/files/Voll ]
+ chmod 0600 /data/data/mattecarra.accapp/files/Voll
chmod: /data/data/mattecarra.accapp/files/Voll: No such file or directory
+ e=1
+ echo
+ exit 1
```

Instead of `/data/data/mattecarra.accapp/files/Voll` it must be `/data/data/mattecarra.accapp/files/Voll aufladen.profile`. Maybe you have to add some quotes to the path?

VR-25, 2019-10-27:
That's an edge case.
Which version are you upgrading _from_?
Wipe AccA's data (or uninstall the app) and try upgrading again.

KreativeKrise, 2019-10-27:
I can't remember, I think it was 1.0.14.. But I also already tried wiping the data and uninstalling the app, which ends with the result which you can see in the logs.

KreativeKrise, 2019-10-27:
Wiping data solved the problem. Thank you @VR-25 :)

closed 2019-10-27 by KreativeKrise