MatteCarra/AccA #76: ACC compatibility with Android 10?

patrickhgb, 2019-11-08:
Hi,

OnePlus 7 Pro GM1917
Android 10 latest stable (install by wiping phone and local upgrade)
Magisk boot image patch root method
ACCA latest version from release APK

I install ACCA by ADB, then I granted superuser rights with Magisk and verified that the daemon is running by restarting it.

In the configuration, I set these values to test;
Shutdown: 3%
Resume: 49%
Start: 79%
Switch: auto
I verified that it works by using the built in test feature, and the phone buzzed to indicate to stop/start charging sequence.

Charging voltage control and Idle mode is not turned on.

The battery percentage is now 59% which should mean that ACCA stops charging and waits for the battery to drop to 49% or below to start charging and stop at 79%.

When I plug in the charger, it actually starts to charge the phone and ACCA is not doing anything to stop it. Status says charging (fast) and speed at -300mA at 3.94v.

Log image: https://i.imgur.com/iUVhJ4j.jpg

I remember ACCA working before, not sure why it broke when I update to Android 10.

Aiakio, 2019-11-28:
Hey,

exactly the same here. Pixel 4 XL. Can't get it to work. It just doesn't stop charging.

MatteCarra, 2019-12-06:
Thanks for the reporting the issue.
This is an acc issue though ( https://github.com/VR-25/acc/issues/ )
Anyway I don't think this is a Androd 10 specific problem, because I'm running acc on OOS 10.
It's very strange

dimqua, 2019-12-11:
ACC works fine for me as well. LOS 17.

VR-25, 2019-12-14:
The 1+7 Pro situation is under investigation.
The Pixel 4 and 4 XL have [3 new](https://github.com/VR-25/acc/issues/20#issuecomment-565711859/) charging switches. 2 of these were confirmed to work.

closed 2020-03-09 by VR-25