MatteCarra/AccA #88: My battery won't charge

JustyouKnow-code, 2020-02-25:
Hello is anyone here know how to fix ACC because when the daemon started my won't charge anymore even I disabled ACC my only way too fix this is to uninstall ACC pls help my ACC is the only app that work my only problem is it won't let me charge my phone when I'm in OS.

VR-25, 2020-03-09:
1. Install the latest ACC from https://github.com/VR-25/acc/releases/
2. Test it

Continue if the issue persists.
Else, close the issue.

3. Run `acc -le`
4. Share `/sdcard/acc-logs-*.tar.gz`

mildred, 2020-04-08:
Hi, I think I'm having a similar issue. On LineageOs 16 klte, since I installed AccA, I cannot charge my phone more than 25~39%. AccA reports a charging state until it reaches a limit and then it reports Discharging (although the charge current approximates 1000mA).

With the latest stable acc (201910132), I executed `acc -l -e` and attached the logs uncompressed here:

- [accd-klte.log](https://github.com/MatteCarra/AccA/files/4449269/accd-klte.log)
- [acc-power_supply-klte.log](https://github.com/MatteCarra/AccA/files/4449273/acc-power_supply-klte.log)
- [charging-ctrl-files.txt](https://github.com/MatteCarra/AccA/files/4449274/charging-ctrl-files.txt)
- [charging-voltage-ctrl-files.txt](https://github.com/MatteCarra/AccA/files/4449275/charging-voltage-ctrl-files.txt)
- [config.txt](https://github.com/MatteCarra/AccA/files/4449276/config.txt)
- [init.log](https://github.com/MatteCarra/AccA/files/4449278/init.log)
- [install.log](https://github.com/MatteCarra/AccA/files/4449279/install.log)
- [install-tarball.sh.log](https://github.com/MatteCarra/AccA/files/4449280/install-tarball.sh.log)

Bortex19, 2020-05-28:
I have the same problem.
[acc-logs-dipper.tar.gz](https://github.com/MatteCarra/AccA/files/4696638/acc-logs-dipper.tar.gz)

JustyouKnow-code, 2020-05-30:
I talk to VR-25 and he try his best to solve my problem but my mediatek device is very troublesome that's why until now I can't still use Acca for anyone who experienced the same problem try to talk to VR-25 in telegram, they have group chat in there. Hope this helps.

VR-25, 2020-07-25:
https://github.com/MatteCarra/AccA/issues/114#issuecomment-663824318

MissyToby2000, 2020-08-04:
> Hello is anyone here know how to fix ACC because when the daemon started my won't charge anymore even I disabled ACC my only way too fix this is to uninstall ACC pls help my ACC is the only app that work my only problem is it won't let me charge my phone when I'm in OS.

I guess I had the very same issue on a Mediatek device (MT6580) and just solved it by installing the latest release, by selecting it via the settings menu. At first, nothing happened, so I rebooted to TWRP, wiped Dalvik and cache. Daemon started, installed Acc and now everything is working as expected. Many thanks to the devs and the OP.

closed 2021-08-03 by VR-25