MatteCarra/AccA #90: BUG REPORT AND FEATURE REQUEST FOR ACCA

mikesew1320, 2020-02-29:
##### FEATURE REQUEST

1)------> Splash Screen (It's very important thing for peoples having lowram devices. The launch of ACCA takes some time on such like devices So we  need a splash screen that contains the cool icon of ACCA. By the way there is a magical splash screen at the current state which is a blank dark background)

2)----> A permanent notification that contains all the required information like battery temperature, voltage current, current acca status, current battery status, charging from....and some other cool variables if you have another much important.

3)-----> Can u lower the font size for the home screen variables like temperature & current now....... please? They becomes appear in two line(Meaning some part of the string Temperature [starting from --ure appear below Temperat---] the same for current now) for devices having lower screen resolution 320ppi 1280×720 devices.

Also at the "About" section the the string  "GitHub"[Found infront of each of the team members] also appears in two line on the same device[GitH--at t
[Screenshot.zip](https://github.com/MatteCarra/AccA/files/4270673/Screenshot.zip)
he top --ub below --GitH]

I think @squabbi should pay attention on this! ACCA has a greater UI related bugs.

##### BUGS

1) On every launch of ACCA, It detects the ACC daemon Status and home screen variables lately after some seconds near 10Sec. It's very annoying! 😡

2) The log pannel doesn't work at all. It doesn't show any log of ACC except "press:q [enter] to quit"

3) Import and export doesn't work at all. Specially the import function. Even can't nask for storage access when I try to export one of the profiles.

##### TELLING YOU WHAT I AM BORED WITH....

1) Can u remove what it is written on the bracket(Telegram link of ACC Group) at the last button on the Settings panel please? It's not necessary at all. Makes the UI too boring.

[SEE THE SCREENSHOT BELOW FOR MORE UNDERSTANDING]

My device is:-

BRAND:-TECNO

MODEL:-Camon CX Air

ANDROID:-7.0 Running on stock HiOS 2.2 ROM

CHIPSET:-MT6737M

RAM:-2GB

ROM:-16GB

GSM ARENA DETAIL :- https://m.gsmarena.com/tecno_camon_cx_air-ampp-9445.php

Thanks!

squabbi, 2020-03-10:
Thanks for the screenshots! We will probably look at making different layouts for different densities so that will minimise the problem.

Mikesew1320, 2020-03-10:
> Thanks for the screenshots! We will probably look at making different layouts for different densities so that will minimise the problem.

Ok, Thanks!! What about the rest? If you need something just ask me here. I will give you the information you need.

squabbi, 2020-07-25:
Update on this issue:

**Feature requests**

1. is finished and implemented here: 58cf9f5164f0aa154d149a7bd299ef8fefeca97d

**Bugs**

1. Fixed in 4dba83b8bad267ca7c6eb5999f0fc335a84ae01f
2. Fixed some time ago (or I cannot replicate it)
3. Fixed by the time this commit was pushed 2581b8108944ee1cbe55e9681c7a1cb07d84c6a1
4. Fixed in 7c67591002aeb6cc3a62d5992162a9e7944e7cfc

**Things you were bored with**

ACCA Telegram 'settings' entry is moved to the About screen: 5c9a52509910d05f41bb5971a4d1d40e499b0c9f

squabbi, 2020-07-25:
Permanent notification will be addressed in #45 

**Feature request**

squabbi, 2020-07-25:
All issues taken care of.

closed 2020-07-25 by squabbi