MatteCarra/AccA #91: Schedule profile name menu is confusing

PacoBell, 2020-03-01:
Right now, that drop-down menu offers the user no information regarding which profile is currently applied for a given schedule. "Keep current config" is rather useless. Just show the actual profile name. Hitting the Cancel button is effectively the same thing as "Keep current config", anyway. Similarly, "Edit current config" would be better implemented as that "pencil" icon you already use elsewhere in the app. It would certainly be more consistent and further simplifies that drop-down.

MatteCarra, 2020-03-01:
"Keep current config" is used to change scheduling time or schedule name.
Thanks for the tips!

PacoBell, 2020-03-14:
Understood, but neither of those changes has anything to do with the config itself. So why have an option available that doesn't ever change state? Eliminate it and replace it with an actual state indicator such as one of the profile names already in that list. Doing so and also replacing the edit menu option with an edit button adjacent to the current config would also provide intuitive context as to exactly what is being edited while allowing more room in the precious drop-down space for potentially more configs without scrolling.