MatteCarra/AccA #101: java.lang.IllegalStateException: Migration didn't properly handle: profiles_table(mattecarra.accapp.models.AccaProfile)

jesse-git, 2020-04-26:
Android 9, LineageOS
ACCA 1.0.23 (27)

Are profiles implemented yet?

```
04-26 17:57:37.458 E/AndroidRuntime(23071): Caused by: java.lang.IllegalStateException: Migration didn't properly handle: profiles_table(mattecarra.accapp.models.AccaProfile).
04-26 17:57:37.458 E/AndroidRuntime(23071): 	at mattecarra.accapp.database.ProfileDao_Impl$9.call(ProfileDao_Impl.java:316)
04-26 17:57:37.458 E/AndroidRuntime(23071): 	at mattecarra.accapp.database.ProfileDao_Impl$9.call(ProfileDao_Impl.java:313)
04-26 17:57:37.472 W/ActivityManager(2533):   Force finishing activity mattecarra.accapp/.activities.MainActivity
04-26 17:57:37.504 I/WindowManager(2533): WIN DEATH: Window{84e7e4b u0 mattecarra.accapp/mattecarra.accapp.activities.MainActivity}
04-26 17:57:37.504 I/ActivityManager(2533): Process mattecarra.accapp (pid 23071) has died: fore TOP 
```

squabbi, 2020-04-28:
Were you upgrading from an older version of the app? We probably left out something during the database migration. Profiles have been implemented and are currently working on 1.0.23 (27) right now, but you will have to clear app data if you were coming from an older build.

If you'd rather keep all your profiles, let us know the previous app version you were on and we'll see to fix the database migration!

jesse-git, 2020-04-28:
Yes but I am not sure which version I upgraded from. It was a while back (months). So I probably skipped some versions. I remember once trying to do profiles with an older version but it didn't work for one reason or another, so I have no old profiles.

After clearing data, will my **acc** version and settings remain? I am running 2020.4.22-beta. Should I backup the acc config or anything?

squabbi, 2020-04-29:
Ah I see. Yes, you should be fine to wipe app data, version and acc settings should persist.

jesse-git, 2020-05-01:
I cleared data, but I think it reinstalled the old version of ACC also.  I had to fiddle with the version selection under settings to make it read the correct version. I'll let you close this when you want

closed 2020-05-10 by squabbi

aradhna2004, 2022-02-28:
But how to migrate database if my app is ready to update without clearing app data?