MatteCarra/AccA #104: Unfortunately, ACCA has stopped.

Gamemulatorer, 2020-06-18:
[image]

### LOGS
[logs-2020-06-18-11-53-53.zip](https://github.com/MatteCarra/AccA/files/4796167/logs-2020-06-18-11-53-53.zip)

Gamemulatorer, 2020-06-20:
@MatteCarra? 😅

MatteCarra, 2020-06-21:
Can you please provide more details?
What were you doing when it crashed?
Were you using the bundled acc or a different version?
Thanks

Gamemulatorer, 2020-06-21:
@MatteCarra thanks for the reply but now I'm using the magisk module of ACC and it's work good on my phone.

dchmelik, 2020-06-30:
I have same problem on an Archos Elements 97 Platinum with an Android 4. Then I select share and it said there's no log...

squabbi, 2020-07-14:
@dchmelik can you provide which version of ACC you are running? Run `acc -v`.

VR-25, 2020-07-25:
A clean install of the latest version from this repository should fix this.

dchmelik, 2020-07-29:
I gave the particular pad/tab to a family member who lives somewhere else.  If they want to use AccA they'll ask me and I'll get back to you...

closed 2021-01-18 by Gamemulatorer