MatteCarra/AccA #105: Acc Daemon is Not Runningg

Georgian-D, 2020-06-18:
Device: Galaxy s7 Edge (G935F)
Android 9 ( custom rom: BlackDiamond.N8.PIE.V9 [https://forum.xda-developers.com/s7-edge/development/blackpearln8stable-smooth-fast-t3827982)

When I enter in the app it shows me the following message up top:
[image]

If I press start it will show like this for a few seconds and it goes back:
[image]

What I am trying to do is to set the battery temperature limit to 30 degrees Celsius, first as a test and it seems that this is not working because right now it reached 39.

Is this normal? If is not, is there a fix?

Thank you :)

MatteCarra, 2020-06-21:
Which version of acc are using?
acc -v

SmokingUnicorn, 2020-06-22:
I have a similar problem. The daemon will start after opening ACCA (without needing to click start) but will stop after some time (sometimes a a few mins up to an hour). Therefore I need to reopen ACCA before charging and auto-shutdown at x% does not function (daemon not running by this time). Any advice much appreciated :)

ACC version: v2020.5.4-rc (202005040) -- This is the version bundled with ACCA

Device HTC U11 (Custom ROM - Leedroid: Oreo V3.5.3-R16. https://forum.xda-developers.com/u11/development/rom-leedroid-v1-0-0-beta-fast-stable-t3623657

VR-25, 2020-07-19:
Fixed in today's acc version.

closed 2020-07-19 by VR-25