MatteCarra/AccA #114: Charging always stops at mid value since latest version

dtrail, 2020-07-21:
Since the last version my device only charges to 70%, then stops and continues charging from 69%, only to stop at 70% again.
I had a working configuration with 15 - 70 - 90, which always worked fine, until the update.
To verify the issue, I've reset the app , cache and data, reconfigured it and also tried a clean reinstallation.
And then I tested the app on another device, with similar results when using the latest version. But there it stopped at 60% although there was no limit set at this value. To confirm that the problem is related to the latest version on ACC daemon, I've set the version to 2020-07-03 (which is the one that works fine for me), but unfortunately on that other device no version except the latest one (or just select "package") is working. The demon won't start , and after a device reboot even the app refuses to start, until I reset it and keep the latest version.

My own device is starlte (Galaxy S9, Exynos)
The other device I've tested the app with is Galaxy Note 4.

I'm pretty sure that the current new version is the reason for this issue.

dtrail, 2020-07-21:
I forgot to pull the logs. I will attach them when I get a chance to do so.

BrommD, 2020-07-21:
Hi all, I can confirm the issue. Device: Samsung S9. Thx

BrommD, 2020-07-21:
Update: this is happening with current ACC, independently of ACCA usage or not. So no ACCA issue, but ACC.

VR-25, 2020-07-25:
Have you tried v2020.7.24?

dtrail, 2020-07-28:
Confirming it's happening with the very latest version too. Switched back to 07.03.

BrommD, 2020-08-01:
check you dropbox, have uploaded an "acc-le", thanks, Dejan

dtrail, 2020-09-11:
The problem is still there. It works for one single time to charge until the given limit (i.e. 80%), but then it always charges only to exactly 70%. It only works again when I reboot my device. Then it charges again to the limit I have set. But after that, again, it will only charge to 70% - until it reboot my phone. And so on...

It happens with both, the app and without the app (when using shell input).

Did I miss something? Is lt also doesn't matter which version I'm using. Is there a way to reset the configuration entirely? Is it on my end, or is it because of my device (S9 starlte)?
Until I find a solution, I'm going to use IPC instead. But that's quite simple and doesn't allow all these functions that ACC has to offer.

VR-25, 2020-10-03:
This was found to be a Samsung-specific issue.
It's caused by the store_mode charging switch.
Upgrade to the latest version and set the batt_slate_mode charging switch with `acc -ss`.
For additional details, refer to readme > troubleshooting > charging switch.

closed 2020-10-03 by VR-25