MatteCarra/AccA #117: Cannot install DJS

gjf, 2020-07-28:
Acca 1.0.26

Pressing scheduler button at the main window gives an alert "DJS should be installed".

When pressing "Install" the same alert appears again and again.

In fact DJS functions are not available.

Hinfty, 2020-08-20:
Had the same issue after everything worked for a while and different version

vilnitsky, 2020-09-05:
Djs installation stopped working too(

adamallo, 2020-09-22:
I have the same issue. I also tried installing DJS directly from Magisk. It installed correctly and can get it's PID, but AccA still has the same issue.

rav101, 2020-09-23:
I have the same issue

Gronkdalonka, 2020-10-01:
Also error upon installing DJS, install script tries to chmod a file which is not there.

[install.log](https://github.com/MatteCarra/AccA/files/5313588/install.log)

Ph0rk0z, 2020-10-17:
me as well, DJS never installs and tries to install again when clicked. djs daemon is checked when I go into settings. "develop" version never installs either even though it appears to download.

sbstratos79, 2020-12-03:
Same issue. Installed DJS with Magisk. Acca still won't let me open the scheduler tab.

arun54321, 2020-12-18:
Same problem. Cannot install djs. Even installed djs from magisk repo. still it doesn't work.

CjiqL, 2021-01-06:
Same problem. On oneplus8t android11

ssorgatem, 2021-02-11:
Same problem on Ulefone Armor 9, even after installing DJS with Magisk

pagdot, 2021-02-18:
If djs is already installed is checked by checking if the installer is there in [Djs.kt#L98](https://github.com/MatteCarra/AccA/blob/develop/app/src/main/java/mattecarra/accapp/djs/Djs.kt#L98). Instead it should be checked if the binary exists using `which djsd > /dev/null` as stated in [djs Readme](https://github.com/VR-25/djs#online-djs-install).

eegeeZA, 2021-02-19:
> If djs is already installed is checked by checking if the installer is there in [Djs.kt#L98](https://github.com/MatteCarra/AccA/blob/develop/app/src/main/java/mattecarra/accapp/djs/Djs.kt#L98). [...]

Good find. You should create a pull request with the changes. Remember to change not only `isDjsInstalled`, but also `initDjs` and `uninstallDjs`, as I see that they all rely on the faulty `installationDir`.

pagdot, 2021-02-19:
Fixed some of the code locally, but couldn't get arsed to setup a proper Android dev setup, tried standalone android-sdk, but had problems with it.

I'll try to get a build later today and submit a PR if I've got it working

MatteCarra, 2021-03-12:
any update @pagdot ?

pagdot, 2021-03-12:
Sorry, I struggled because I didn't get djs to working when I tried it. I need to test it more. Additionally I need to make sure to call the initi script if necessary. I made a few mistakes at my first try. I try to continue this weekend

blockberd, 2021-06-13:
Any update on that topic? I just found out about AccA and really want to use the DJS feature. I did the same things as above commenters

pagdot, 2021-06-14:
I had issues that djs worked, but the command interface of AccA didn't work for me somehow most of the time.

scrouthtv, 2021-06-16:
Maybe this is related to this DJS/A11 issue: https://github.com/VR-25/djs/issues/8

pagdot, 2021-06-16:
No, I verified that DJS works by creating/modifying files with a djs job. I don't remember if I tried executing the acca commands myself, but I think I did and they didn't work either

scrouthtv, 2021-06-17:
On Android 11?

nortuzar, 2021-07-03:
Same problem here with ACCA 1.0.31, try all combinations but nothing works. Is there a manual fix to this?
Thanks for the great app

MatteCarra, 2021-08-02:
Fixed.

closed 2021-08-02 by MatteCarra