MatteCarra/AccA #120: Scheduled profile changes are not executed

Hinfty, 2020-08-05:
I have installed DJS via AccA (and also tried via Magisk), but the scheduled profile changes work unreliably during deep doze.
During (nightly) charging, the profile change works fine.
But when I switch to daily profile later in the morning without charging, its unreliable. 

This could be due to DJS or Acc, not obvious. 
Is there a way to exclude Acc or DJS from battery optimisation?
How does DJS behave during deep doze?

VR-25, 2021-08-03:
ACC and DJS run detached from the Android framework.
Thus, doze should not affect them.
Try the latest releases and report back.

blockberd, 2021-08-08:
I am on the latest version of Acc, AccA and DJS but the DJS does not switch to the proper profile when the time is reached. I have 2 schedules. One for charging up to 80% during night and one to top off to 100% in the morning. Both schedules use profiles previously created (ie. not a profile directly setup within the schedule).

VR-25, 2021-08-08:
> I am on the latest version of Acc, AccA and DJS but the DJS does not switch to the proper profile when the time is reached. I have 2 schedules. One for charging up to 80% during night and one to top [...]

Share `/data/adb/vr25/djs-data/config.txt`

blockberd, 2021-08-08:
I hope this works and is what you require.

https://github.com/MatteCarra/AccA/files/6950392/config.txt

VR-25, 2021-08-08:
> I hope this works and is what you require.
> 
> https://github.com/MatteCarra/AccA/files/6950392/config.txt

This is it, thanks.

> Edit: I would like to point out that I have not done any modifications to that file on my own manually. If there are settings incorrect, they must have come through a UI.

Got it.
It seems djs needs some optimizations to handle a few exceptions.
AccA's source code also requires some tweaking.
Will work on those.

VR-25, 2021-08-10:
There's a [new](https://github.com/vr-25/djs/releases/latest) djs release.
Try it an report back.
Thanks

blockberd, 2021-08-10:
I did some basic / simple testing and it seems to switch correctly between the different profiles. One thing I noticed:
It seems to change the charging limit values (visible on the main screen) but the actual profile is not being marked with the blue bar on the left in the profile screen. So it seems to get the correct values from the profile (which is fine) but its not visually visible within the profile screen (which is not a big problem). Just wanted to mention that.

It cleary did NOT change the charging limit values prior to the DJS upgrade today. So that solved the issue. Thank you.

VR-25, 2021-08-10:
Thanks for the collaboration!

closed 2021-08-10 by VR-25