MatteCarra/AccA #123: (request)

droid-jd, 2020-08-13:
It's just a minor inconvenience, and it likely doesn't bother anyone other than me. Truthfully, NOT a big deal in the slightest- the app is great!

Basically, when the menu pops up after opening " charge once to xyz% without restrictions", I keep finding myself tapping slightly off from "apply", and I don't realize that the menu closes and my charging current remains unchanged. Just wondering if it's possible to menu from closing if you tap outside the menu box. 

Like I said, not at all a big deal, just as minute polish point. Everything that the you and the other developers do here is amazing.

squabbi, 2020-08-20:
I can make the dialog non-dismissable, or perahps create a little bit more padding around the "Apply" button!
I'll also change the toast message to something with more details too.

Thank you for the suggestion!

droid-jd, 2020-08-20:
Awesome, thank you so much.
I know it's a minor issue, but I think it tidies things up a bit.
LOVE the app and what you and the other devs have done with it.

commit 1072d948 (2020-09-07): Make Charge once to without restrictions dialog non-dismissable by touch.
Adjust string resources used to show if action has completed successfully, or dialog is dismissed.

Closes #123
closed 2020-09-07 by squabbi