MatteCarra/AccA #130: Possible errors or is my phone not supported?

FootballFan141, 2020-09-20:
I downloaded the Advanced Charging Controller magisk module and seem to be having trouble with AccA.
I am running the Havoc OS ROM with the latest haumea kernel on LG V30

I attached the logs with 3 screenshots

Screenshot 1: Shows battery temperature as -1 degrees
Screenshot 2: Says DJS is required but I have it installed..
Screenshot 3: What shows when I click the logs button in the app. (Thought it could be helpful)

If anyone thinks I would have better luck at XDA or Reddit, just let me know.

[acc-logs-joan.zip](https://github.com/MatteCarra/AccA/files/5251114/acc-logs-joan.zip)

squabbi, 2020-09-24:
Hi, DJS probably needs a bit of a touch-up in ACCA. As for the others I'm not too sure. I can let the devs know of the issues to ACCA and ACC.

MatteCarra, 2021-08-02:
Please test with latest ACCA version.

VR-25, 2021-08-10:
@FootballFan141, have you tried the latest acc?
Could you provide a fresh log archive right after running `acc -i` once?