MatteCarra/AccA #131: Acca doesn't install on Android 11

adri-k, 2020-09-21:
I am trying to install Acca version 1.26 on my Pixel 4, running Android 11.
After installing I get a message 'App not installed'.
When I try to install the older version 1.24, the installation is successful, but as soon as I try opening Acca, I briefly see the main screen and then it crashes.
When I installed version 1.24, using adb I see there is no folder /sbin, which is supposed to hold the acc executables.

leifalbor, 2020-09-21:
Same issue here.  v1.0.26, Pixel 2 XL, Android 11 (fresh install).

crsag, 2020-09-23:
I'm also having same issue here. v1.0.26, Pixel 4 XL, Android 11 (fresh install). acc is working fine though. But when I tried to install front end application from fdroid 
I gets  "Error installing Advanced Charging Controller (ACCA) Failed to install due to an unknown error".

adri-k, 2020-09-23:
@crsag,

How did you install and configure acc?
The zip-file is supposed to be flashed from recovery, but there is no TWRP for a Pixel 4.
Installing only acc might be a temporary solution for me.

crsag, 2020-09-23:
> @crsag,
> 
> How did you install and configure acc?
> The zip-file is supposed to be flashed from recovery, but there is no TWRP for a Pixel 4.
> Installing only acc might be a temporary solution for me.

From Magisk manager

adri-k, 2020-09-23:
Thanks.

I'll try that.

MatteCarra, 2020-11-13:
Please reopen if the issue has not been resolved

closed 2020-11-13 by MatteCarra