MatteCarra/AccA #135: Changes to profile not sticking

jordanade, 2020-10-04:
I change settings and hit the save icon. When I check they are saved, but later they revert back to the default.

MatteCarra, 2020-11-13:
Thanks for the report.
Which changes are you trying to apply?
Be specific so I can reproduce.
Thanks a lot

andiwirs, 2020-12-24:
I can confirm that issue. For me for example the max. temperature does not stick.
The default when i installed the app was 60°C. I changed it to 45°C hit save, closed the app, reopened, had a look at the settings and it was at 60°C again.

The procedure that helped for me was to change (in my case) the "default" profile to my liking and then apply the profile...

jordanade, 2020-12-24:
Yep, same thing here. It works if I make sure to only edit and apply a
profile instead of the thing that pops up on the home screen.

MatteCarra, 2021-08-02:
I can not reproduce the issue on latest version (from github).
Please test and close the issue if fixed

andiwirs, 2021-08-03:
For me it seems to work now. See also #156

closed 2021-08-10 by VR-25