MatteCarra/AccA #142: [Feature Request]: battery idle mode

imanmoghimiq30, 2020-12-04:
Hello
Thanks for great app.
I work in company like Uber as driver and use my phone as navigator and accept customer requests. Also many people use their phone as navigator and infotainment in car.

So if we have button in notification center simply active battery idle mode and ignore all profile would be good in above situations.
I know I can do this with a custom profile but I think it's more user friendly. 
Regard 
Be safe
Iman from Iran

VR-25, 2021-08-03:
@MatteCarra, I believe this is simple for you to do, using just `acca --disable` and `accd` commands.
The former accepts arguments: `#[%hms]`, e.g., `acca --disable 90%` (until level drops to 90%), `acca --disable` 10s (for 10 seconds), `acca --disable 3h` (for 3 hours).
With those arguments, we don't need a timer in AccA.

VR-25, 2021-08-10:
We could have custom buttons to run acc commands/presets on demand.