MatteCarra/AccA #152: acca 1.0.29 install acc failed on android 4.1.1 and

CrazyBoyFeng, 2021-02-21:


CrazyBoyFeng, 2021-03-06:
`acca.apk` let me upload `/data/adb/acc-data/logs/`, but it was not existed, and I found `/sdcard/vr25/acc/logs/install-tarball.sh.log`
[install-tarball.sh.log](https://github.com/MatteCarra/AccA/files/6095954/install-tarball.sh.log)

VR-25, 2021-03-07:
It seems to be a busybox problem.
That version might be too old.
You can simply download the appropriate busybox binary and place it in /data/adb/bin (permissions set to 700 or 755).
Magisk's and Osm0sis' busybox are great.
You can find those here on GitHub.

CrazyBoyFeng, 2021-03-07:
I used meefik/busybox 1.31.1 before, whose version is the same as osm0sis just because it is easy to install.  
I found acc's install log said it located `busybox` at `\system\bin\` but I clearly install it to `\system\xbin`, I think this may be the cause of the error.  
So I `ln /system/xbin/busybox /system/bin/busybox`, and start acca, failed either. `/sdcard/vr25/acc/logs/install-tarball.sh.log`:
```
18: [ -x /dev/.busybox/ls ]
19: mkdir -p /dev/.busybox
20: chmod 0700 /dev/.busybox
21: [ -f /data/adb/bin/busybox ]
24: [ -f /data/adb/magisk/busybox ]
27: which busybox
27: >/dev/null 
0: which busybox
28: eval /system/xbin/busybox --install -s /dev/.busybox
0: /system/xbin/busybox --install -s /dev/.busybox
36: export PATH=/data/adb/bin:/dev/.busybox:/sbin:/vendor/bin:/system/sbin:/system/bin:/system/xbin
0: id -u
41: [ 0 -ne 0 ]
46: umask 0000
47: set -e
50: [ -f //install-tarball.sh ]
0: readlink -f /data/data/mattecarra.accapp/files
50: cd /data/data/mattecarra.accapp/files
68: trap copy_log EXIT
71: rm -rf acc[-_]*/
71: 2>/dev/null 
72: tar -xf acc_bundle.tar.gz
0: get_ver acc_bundle.tar.gz acc_v2020.10.15_(202010150)
77: sed -n /^versionCode=/s/.*=//p acc_bundle.tar.gz/module.prop
77: 2>/dev/null 
77: echo 0
78: bundled_ver=0
0: get_ver /data/adb/vr25/acc
77: sed -n /^versionCode=/s/.*=//p /data/adb/vr25/acc/module.prop
77: 2>/dev/null 
79: regular_ver=202010080
79: 2>/dev/null 
```
Then I think maybe the meefik's busybox is incompatible, so I follow your instructions, install Magisk-Modules-Repo/busybox-ndk to `/data/adb/bin/busybox`, then I start acca, failed either, and the `install-tarball.sh.log`:
```
18: [ -x /dev/.busybox/ls ]
36: export PATH=/data/adb/bin:/dev/.busybox:/sbin:/vendor/bin:/system/sbin:/system/bin:/system/xbin
0: id -u
41: [ 0 -ne 0 ]
46: umask 0000
47: set -e
50: [ -f //install-tarball.sh ]
0: readlink -f /data/data/mattecarra.accapp/files
50: cd /data/data/mattecarra.accapp/files
68: trap copy_log EXIT
71: rm -rf acc[-_]*/
71: 2>/dev/null 
72: tar -xf acc_bundle.tar.gz
0: get_ver acc_bundle.tar.gz acc_v2020.10.15_(202010150)
77: sed -n /^versionCode=/s/.*=//p acc_bundle.tar.gz/module.prop
77: 2>/dev/null 
77: echo 0
78: bundled_ver=0
0: get_ver /data/adb/vr25/acc
77: sed -n /^versionCode=/s/.*=//p /data/adb/vr25/acc/module.prop
77: 2>/dev/null 
79: regular_ver=202010080
79: 2>/dev/null 
```

SmallAndSoft, 2021-03-29:
acc install fails for me too on Nexus 5 Lineage OS 14.1 (Android 7.1.2) with Magisk busybox 1.31.1
AccA reports that daemon is running even though logs sat that acc is not found.
AccA crashes when I switch to profiles tab.

VR-25, 2021-08-03:
How about the latest acc release?

CrazyBoyFeng, 2021-08-03:
Thanks for your work. I will take the time to test it. Then report back by the end of this week.

CrazyBoyFeng, 2021-08-06:
Result: Install failed
AccA version: 1.0.33 from GitHub
Device: ZTE U970 (The previous Android 4.1.1 was Vido N70S. That device was not with me recently.)
Android: 4.1.2
Kernel: 2.6.39.5
Root: SuperSU
Busybox: 
* Magisk-Modules-Repo/busybox-ndk 1.33.1 at /data/adb/bin/busybox with permission 0755
* meefik/busybox 1.31.1 at /system/xbin/busybox

[logcat.txt](https://github.com/MatteCarra/AccA/files/6946312/logcat.txt)
[/data/adb/vr25/acc-data/logs/install.log](https://github.com/MatteCarra/AccA/files/6946327/install.log)
[/data/adb/vr25/acc-data/logs/install-tarball.sh.log](https://github.com/MatteCarra/AccA/files/6946329/install-tarball.sh.log)

VR-25, 2021-08-06:
According to the logs, the issue is related to busybox, `chcon: not found`.
The question is, how coe a recent busybox build not include `chcon`?
I'll be investigating this further.

VR-25, 2021-08-06:
Move /data/adb/bin/ to /data/adb/vr25/ and try again.
You should have the path `/data/adb/vr25/bin/busybox`.

CrazyBoyFeng, 2021-08-06:
I moved `/data/adb/bin/busybox` to `/data/adb/vr25/bin/busybox`, then `chcon: not found` either:
[install.log](https://github.com/MatteCarra/AccA/files/6946788/install.log)
[install-tarball.sh.log](https://github.com/MatteCarra/AccA/files/6946789/install-tarball.sh.log)

VR-25, 2021-08-06:
Does `/dev/.vr25/busybox/chcon` exist?
Either way, run `echo > /dev/.vr25/busybox/chcon` and try installing acc again.

CrazyBoyFeng, 2021-08-06:
`/dev/.vr25/busybox/chcon` does not exist.
[ls /dev/.vr25/busybox](https://github.com/MatteCarra/AccA/files/6946856/ls.busybox.txt)

I tried `echo > /dev/.vr25/busybox/chcon`, then `chcon: not found` either
[install.log](https://github.com/MatteCarra/AccA/files/6946873/install.log)
[install-tarball.sh.log](https://github.com/MatteCarra/AccA/files/6946876/install-tarball.sh.log)

CrazyBoyFeng, 2021-08-06:
Does the `chcon` require `selinux`?

VR-25, 2021-08-06:
```
#!/system/bin/sh
# dummy chcon
# place me in /system/bin, /system/xbin or any directory that is in PATH
# set proper permissions
exit 0
```

CrazyBoyFeng, 2021-08-06:
Now:
`Magisk-Modules-Repo/busybox-ndk/busybox-arm` 1.33.1 at `/data/adb/vr25/bin/busybox` with permission `0755`
Dummy `/system/xbin/chcon` with permission `0755`
`/dev/.vr25/busybox/chcon` does not exist.

Try install failed.
`install.log` does not exist.
[install-tarball.sh.log](https://github.com/MatteCarra/AccA/files/6947149/install-tarball.sh.log)

By the way, after running `echo > /dev/.vr25/busybox/chcon`, `install.log` doesn't exist either, sorry for the mistake.

CrazyBoyFeng, 2021-08-06:
After rebooting, there are some files more. Since GitHub doesn't support some of these formats for upload, I packaged them:
[logs.zip](https://github.com/MatteCarra/AccA/files/6947221/logs.zip)

VR-25, 2021-08-06:
> I tried reinstall after rebooting, this time it seemed the `acc` installed successful: [logs.zip](https://github.com/MatteCarra/AccA/files/6947221/logs.zip)

Great!
I took a look at the logs.
Things seem right this time.
Now you only have to deal with the lack of known charging switches for this device.
`acc -p` is a nice tool for that.
It generates a list of potential charging switches.
You'll have to look at each and test those that [remotely] resemble a charging switch.
I'll have a closer look at the power supply log later and provide some suggestions if I find anything relevant.

> But `acca` crashed: [logcat.txt](https://github.com/MatteCarra/AccA/files/6947246/logcat.txt)
> 
> `acca` have some compatibility problem with the API level 16. I know a little about android app development, I'll try to debug it if I have time.

That's also a job for @MatteCarra  and @squabbi.
Perhaps there's room for me too.
Will see...

> Do you think the current issue is considered solved? If you don't think you need to follow up, either close it.

Yes, it's solved.
The next acc will recognize this and automatically _emulate_ chcon.

Thanks for the collaboration!

closed 2021-08-07 by CrazyBoyFeng

CrazyBoyFeng, 2021-08-21:
I spent some time discovering that acca's crash on api 16 seemed to be related to some dependency library. Due to the difficulty of modifying the dependency library, and the complexity of this project, I gave up on finding the cause of the problem. I will try to write a new, simple app that runs for the old device.

VR-25, 2021-08-21:
> I spent some time discovering that acca's crash on api 16 seemed to be related to some dependency library. Due to the difficulty of modifying the dependency library, and the complexity of this [...]

Looking forward to this.
Contact me if you need any assistance/tips regarding the use of acc as a back-end.