MatteCarra/AccA #156: Can not apply changes with ACC v202102250

andiwirs, 2021-02-28:
Hello, i updated ACC through Magisk (Module) to the latest version v202102250.
Now i can not apply changes anymore. If I go to settings, change something and hit save i get back to the main screen.
If i want to check if the settings were applied and go to settings again the screen turns black and AccA crashes after some time. After restarting AccA and checking the settings I see that my changes were not applied. Same thing happens if i try to load a profile...

jesse-git, 2021-02-28:
Me too. Then I tried to uninstall ACCA and acc completely and go back to older versions, but the same issue now plagues all versions which I don't understand

Trying to apply changes as you describe, for me , now causes ACCA to no longer update the current battery stats on the first tab, a sign that something froze up. A force close seems to fix it, but then the problem happens again as soon as trying to apply any change again.

andiwirs, 2021-03-01:
I have not tried an older version (yet). One thing i could imagine is, that it already didn't work anymore before the update of ACC but i haven't noticed. One possible explanation in my case is that i updated to Android 11 in the mean time.

mbielak012, 2021-03-01:
ACCA not support latest magisk module of ACC. So if you use app, uninstall latest magisk module, then restart Device. After restart Let the app install itself magisk module, then restart again (Master).

andiwirs, 2021-03-03:
Thanks for the hint, but unfortunately does not work. AccA is not able to install Acc and errors out. Guess i just wait for the general compat. update of AccA and so long use the terminal wizard instead

SmallAndSoft, 2021-03-29:
Stopped working for me too. Such a shame!

jordanade, 2021-06-03:
Yeah I'm stuck with a broken ACCA as well. Is there some configuration file that needs to be erased or something?

VR-25, 2021-08-03:
Have you guys tried again after a config reset?
`rm /data/adb/vr25/acc-data/config.txt`

andiwirs, 2021-08-03:
Uninstalled ACC updated ACCA to the latest version and let it install ACC again. Seems to work now. 
Thanks a lot.

closed 2021-08-03 by andiwirs