MatteCarra/AccA #167: Cannot change profiles in AccA

bege10, 2021-06-01:
I installed AccA 1.0.29 from F-Droid, then updated ACC module to 2021.2.25 in Magisk 23001.
The daemon is running but I cannot switch profiles. Tapping on a profile shows toast that the profile has been chosen but it never becomes active.
Restarting the phone doesn't help.

[acc-logs-FP3.tar.bz2.zip](https://github.com/MatteCarra/AccA/files/6577424/acc-logs-FP3.tar.bz2.zip)
[logs-2021-06-01-16-27-39.zip](https://github.com/MatteCarra/AccA/files/6577425/logs-2021-06-01-16-27-39.zip)

jordanade, 2021-06-03:
Yeah there's no selected indicator showing for profiles, and after choosing one, clicking the gear icon on the main page leads to an app freeze. This continues even when downgrading to an older version and rebooting.

kgursu, 2021-06-29:
Similar or possibly the same here. Do you need anything from us to fix?

navinthomas, 2021-07-23:
I have got the exact same issue. I get a toast message that the profile has been switched. But, when I click the gear icon, the app freezes. I have to either, force stop the app, or clear it from recents, to make it work again. The profile doesn't seem to change after all that.

MatteCarra, 2021-08-02:
I am not able to reproduce the issue with latest version (on github).
Can you check if it's fixed?

MatteCarra, 2021-08-02:
I will publish a version on fdroid soon, but i want to fix a couple of things first

navinthomas, 2021-08-02:
Tested the latest version (v1.0.32) that is available from GitHub. Issue no longer reproducible for me on my device.

I am able to change profiles without any noticeable issues. On changing the profile:

- The toast message appears
- Within a few seconds, the profile change is indicated by the app by highlighting the selected profile
- After going to the main screen, and clicking the gear icon, I am able to see the updated settings without the app freezing.

Thanks for the fix, Matteo.

Device: OnePlus One (bacon/A0001)
Android Version: 10 (LineageOS 17.1)

closed 2021-08-02 by MatteCarra

nialone, 2021-08-03:
Hi everyone. The problem is not in ACCA, the problem is in the ACC service, after sending the command: "D/SHELL_IN: /DEV/ACCA --SET --Current -". My phone can go to a reboot in 80% or ACCA cannot get configuration and freezes. If you kill the application and run again: new settings are read. There are no problems on the emulator, only on the real device. Accordingly, in the ACCA application, if you select a profile or edit profile and save ... The problem is reproduced stably.

I do not use current settings, so in my assembly disabled the shipment in the code. ConfigUpdater.kt: line 19

I do not know yet you need to display separate settings for sending commands to ACCA. It will be difficult for users who have not read a help. Postponed the idea of ​​the box.

[image]
[image]