MatteCarra/AccA #168: Poco F1 not charging after uninstalling ACC.

AlippRosli, 2021-07-12:
Hi, I'm using Poco f1 android 11 with pixel experience rom running ACC v2021.2.25 (202102250). After installing ACC through my magisk app, this module gave me some problems like charging but my battery logo said its not.  I have tried to restore to default settings and uninstall it but my phone could not charge anymore. I need to reinstall this module and ACC to make it start charging again. Can anyone help me? What should I do to completely restore my settings and uninstall it?

DerpSpears, 2021-07-15:
I would recommend a reinstall of your rom.

VR-25, 2021-07-21:
This must be related to the PMCI bug described [here](https://github.com/lybdroid/poco-x3-pro/issues/1) and [here](https://forum.xda-developers.com/t/rom-official-arrowos-11-0-android-11-0-vayu-bhima.4267263/post-85119331).
Hence, it's not and acc issue.
I'll add this information to the documentation.
Thanks.

closed 2021-07-21 by VR-25