MatteCarra/AccA #177: Allow ANY capacity value, while still ensuring RESUME is always lower than PAUSE

VR-25, 2021-08-11:
I implemented a new acc feature that lets one use millivolts as opposed to capacity/level thresholds.
The new algorithm kicks in **only** when the given value is >= **3000** (mV).

This is useful especially for devices that don't provide reliable capacity measurements (I've got one).
It's also another alternative to voltage limit - which works only on very few/special devices.

One can have some capacity thresholds be in millivolts and others in %, **simultaneously**.
This means, acc 4200 20 works just as well as acc 80 75.
By the way, all capacity thresholds, including shutdown_capacity can be in millivolts.

nialone, 2021-08-11:
In what version are these new ACC functions?

VR-25, 2021-08-11:
> In what version are these new ACC functions?

[v2021.7.28 (202107280)](https://github.com/VR-25/acc/releases/tag/v2021.7.28) and newer

> This is necessary to switch graphic interface.
> How to send a new command in the terminal? You recorded in the FAQ? ))

All the information is in [README.md.](https://github.com/VR-25/acc/blob/dev/REmdADME.)
Read [NOTES/TIPS FOR FRONT-END DEVELOPERS](https://github.com/VR-25/acc#notestips-for-front-end-developers), [Terminal Commands](https://github.com/VR-25/acc#terminal-commands) and [DEFAULT CONFIGURATION](https://github.com/VR-25/acc#default-configuration).

CrazyBoyFeng, 2021-09-05:
If `resume` is set to 80 and `pause` is set to 3200, it is invalid, right? So we need to make them in the same units, right?

The two-unit format is a bit complicated, both for our program and for the user. Can this solution be optimized any further?
Does the conversion formula `capacity=(voltage-3000)/(4200-3000)` apply to all devices? If so, we can use voltage internally and then show capacity to the user.
If it doesn't apply to all devices, then we should probably make a database. Automatic internal conversion for such few devices, and show capacity to the user.

VR-25, 2021-09-05:
> If `resume` is set to 80 and `pause` is set to 3200, it is invalid, right? So we need to make them in the same units, right?

No.
It works.

Quoting myself...
_One can have some capacity thresholds be in millivolts and others in %, simultaneously.
This means, acc 4200 20 works just as well as acc 80 75.
By the way, all capacity thresholds, including shutdown_capacity can be in millivolts._

> The two-unit format is a bit complicated, both for our program and for the user. Can this solution be optimized any further?
> Does the conversion formula `capacity=(voltage-3000)/(4200-3000)` apply [...]

The logic applied depends on the value given.
Any capacity value above 3000 is considered millivolt as opposed to %.

CrazyBoyFeng, 2021-09-05:
When `resume` is set to 80 and `pause` is set to 3200, the `pause` voltage converted to capacity is 17. I say it's invalid to emphasize that it doesn't meet the `resume<pause`.
In this example, to ensure that `resume<pause`, we need either convert or keep the units same.

VR-25, 2021-09-05:
I totally missed your point.
My bad!
Indeed that's an issue.
However, I would assume that a somebody who intends to use % and voltage simultaneously would know that 3200 is too low.

CrazyBoyFeng, 2021-09-05:
I just learned that the formula I was using was incorrect. The power-voltage is not a linear relationship. And there is no universal formula for devices. So to avoid abnormal usage, we need to make sure the units are same.
Or we don't open this feature to normal users and restrict it to advanced users. For example, provide a tougle for advanced features. In advanced mode, no restrictions are on the user input. Otherwise `resume=3200` and `pause=80` may be incorrectly considered invalid. They are valid, but do not meet the&nbsp;`resume<pause`.

VR-25, 2021-09-05:
An _advanced user_ mode is a great idea.
Any values out of the [-1-101]% range should fire a warning notification.

nialone, 2021-09-06:
Personally, I do not understand the battery and must confuse the value. It is best to limit 2 modes: only percent OR millivolts. Ideas? Where are you going to use your advanced mode? It is not comfortable.