MatteCarra/AccA #179: [Feature Request] detect wake up time

ristein, 2021-08-11:
The app could detect the time the user wants to wake up in the morning from set alarms and give an option to be at X% at this time

Dark-Noir, 2021-09-18:
+1 also a sleep as android integration would be nice, when possible.

nialone, 2021-09-18:
Example? I want a lot of examples! ) Do you want a phone that knows how to quickly short-circuit battery or instantly gain percentage?

tamas646, 2022-12-25:
I think it would be a great feature.

Here is a possible implementation that came into my mind:

In the schedule popup there could be options for alarm based time (e.g. '30 minutes before alarm', '1 hour before alarm', 'at alarm' or maybe 'X minutes before alarm') and the app could set the next schedule in djs when:
- the schedule is saved
- an alarm is set or changed or deleted (I don't know whether there's an event for these or not)