MatteCarra/AccA #181: Error when installing

iPot, 2021-08-16:
So I granted the app root access to install something but I got an error like the one on the screenshot.

> [image]

[install-tarball.sh.log](https://github.com/MatteCarra/AccA/files/6994196/install-tarball.sh.log)
[init.log](https://github.com/MatteCarra/AccA/files/6994197/init.log)
[install.log](https://github.com/MatteCarra/AccA/files/6994198/install.log)
[logcat.txt](https://github.com/MatteCarra/AccA/files/6994200/logcat.txt)

VR-25, 2021-08-17:
According to the logs, the back-end (acc) was installed successfully.
The app should be able to detect that acc is installed.
Have you tried launching AccA again after clearing it from recents?

P.S., running `su -c acc -v` on terminal should print acc's version.

iPot, 2021-08-18:
The app works normally. I only use it for the pause and resume charging.
It's just that that error appears when I install the app for the first time but it goes to its menu like everything is normal.

closed 2021-08-18 by iPot