MatteCarra/AccA #183: ACC daemon randomly stops

v1993, 2021-08-24:
I'm not sure what information may be useful to you, so please tell me what I've missed.

I'm using AccA from F-Droid while having ACC installed via Magisk, on Huawei SCL-L21 with lineage OS 14.1 installed. The problem is that ACC daemon appears to randomly stop time to time, something that didn't happened in slightly older versions. Both AccA and ACC are updated to latest available versions. Logs in AccA don't appear to feature anything of interest.

VR-25, 2021-08-25:
Right after the daemon stops, the output of `acc -l tail` would be helpful.
Thanks.

v1993, 2021-08-25:
Is it a problem that I have ACC installed via magisk? Does AccA pick it up automatically?

I'll try to replicate this issue and send output.