MatteCarra/AccA #189: acca kinda bugged on latest update

kevin01523, 2021-08-31:
[image]
[image]
[image]

i think this related to the latest update of acc which kinda broke both acca and acc

hyperfekt, 2021-08-31:
I can confirm this issue occurs on my S10e with LineageOS 18.1 only with the latest release.

VR-25, 2021-08-31:
https://github.com/VR-25/acc/issues/113

VR-25, 2021-08-31:
@MatteCarra, @nialone, this seems to have been introduced by a pull request.
Someone just confirmed that the previous AccA version does not have this issue when paired with the latest acc.

MatteCarra, 2021-08-31:
Can you please confirm me that the only problematic screen is the acc settings screen?

nialone, 2021-08-31:
It is difficult for me to disassemble all the problems because of the language barrier. I realized that there are 2 problems, it is a current and configuration editor. In my last personal assembly all the work!

I can confirm that the editor has already been corrected.
He expects to branch on a confirmation.

[Fix AccEditor](https://github.com/nialone/AccA/commit/82ac445ac4187b0c6071e4d91866784bab3fae41)

kevin01523, 2021-08-31:
yes the only issue is config editor  on the dashboard and on the profiles tab

VR-25, 2021-08-31:
I fixed the 0 mA current.
A new acc version will be pushed once you guys show me a green light.
[acc_v2021.8.31_(202108310).zip](https://github.com/MatteCarra/AccA/files/7084557/acc_v2021.8.31_.202108310.zip) @kevin01523, this should work with the previous AccA.
[acc_v2021.8.31_(202108310).tar.gz](https://github.com/MatteCarra/AccA/files/7084556/acc_v2021.8.31_.202108310.tar.gz)

closed 2021-08-31 by MatteCarra

kevin01523, 2021-08-31:
thanks it works now just needs to test if there's more bugs