MatteCarra/AccA #196: Not yet working on Pixel 6 Pro?

rschuech, 2021-11-03:
Just got a Pixel 6 Pro and rooted with Magisk 23012 but AccA doesn't seem to work.  The built in log just says
/system/bin/sh: <stdin>[112]: acc: inaccessible or not found

This error seems to come up for any ACC version I select in the app, and it says ACC Installation Failed.

I can't seem to easily find the mentioned /data/adb/acc-data/logs folder on the phone (using a root file browser).

When I try to Start the ACC Daemon, it does say the app was granted superuser permissions, but then it soon stops running.

I'm very interested in using this app - it's the main reason I rooted!  Thanks for any help!

VR-25, 2021-11-03:
Magisk 23012 is broken and there are no known working charging switches for Pixel 6 Pro yet.
ACC ships with a pair of recently discovered switches.
However, tests suggest that they are not currently working.
This is under investigation.

Meanwhile, refer to [Current and Voltage Based Charging Control](https://github.com/VR-25/acc#current-and-voltage-based-charging-control).

To install acc without Magisk manager, extract the zip and run `sh /path/to/install.sh` as root.

VR-25, 2021-11-03:
A new acc release is up.
It should install.

rschuech, 2021-11-03:
Yes, I updated and it works!  Thanks SO much for the ultra rapid replies and fix!!

rschuech, 2021-11-04:
It would seem I spoke too soon.  I did not actually do much with AccA last time, and it turns out that the daemon now only runs slightly longer (a minute?) before stopping again.  The log is below.  Using v2021.11.3.  Thanks for any help!

124: true
124: not_charging
81: dis_not 'dis/not' 
45: test-f 'sm????_bms/status'
48: grep -Eiq 'disinot' battery/status
125: return 7
125: exxit
89: exitCode=7
90: false 
90: set +eux

rschuech, 2021-11-05:
I'm not sure what I changed if anything but the daemon seems to stay running now.  I set the charging switch to 0 and disabled auto cycle through switches.  I'm experimenting with the max voltage and max current.  Really noob question - what are reasonable values or ranges for each of these?  500 and 3920?  What the smallest values be that still allow the phone to charge?

VR-25, 2021-11-05:
Exit code 7 means _all charging switches failed to respond_ or _there are none_.
The daemon stops, since there's no point in insisting.

A current of 500+ mA will allow the battery to charge (less = slow).
Any voltage limit will allow the battery to charge up to that threhold.
3920 millivolts is said to be the sweet spot for longevity.
However, it may not offer much capacity for everyone.
The battery may charge only up to about 60% at most.
Try 4100-4200mV to get higher capacity.

At the begining of this issue, I sugested reading  [readme > tips > current and voltage based charging control](https://github.com/VR-25/acc#current-and-voltage-based-charging-control).
Those are alternatives to charging switch based charging control.

VR-25, 2021-11-06:
`acc -t /sys/devices/platform/google,charger/charge_stop_level 100 5`
If that command reports that the switch works, run the following to use the charging switch:
`acc -s s="/sys/devices/platform/google,charger/charge_stop_level 100 5"`

closed 2021-11-06 by VR-25