MatteCarra/AccA #199: Installation failed on Nexus 9 with Magisk and MicroG

kommserv, 2021-11-18:
I installed the [app from F-Droid](https://f-droid.org/packages/mattecarra.accapp/). 

On first start of the app, superuser prompt displayed as expected and I granted rights "forever".
The app then proceeded with "ACC installation...", and shortly thereafter I got the error message **Installation failed** with instructions to open a ticket.

I have copied the [contents of the logs directory and uploaded it to my Nextcloud](https://nextcloud.asks.se/s/wq58FDEYKwSbdDe) (note, I have extracted the contents of the tarball `acc-logs-flounder.tar.gz`).

I'm trying to repurpose an old tablet into a webcam (using Iriun) and extra screen real-estate, so it's more or less permanently connected to power via USB to computer, and I thought the battery could benefit from a set charging limit.

Nexus 9 running LineageOS `14.1-20211021-unofficial-flounder` (Android `7.1.2`) with Magisk `23.0`, MicroG NoGoolag `2.9` with NanoDroid Patcher `23.1.2.x`, TWRP `3.5.1_9-0`, and Xposed framework `89` with XPrivacyLua `1.27`.
API level 25, kernel version `3.10.103` (razorloves), SELinux stats enforcing.

Any ideas on what could be going wrong?

VR-25, 2022-01-29:
Have you tried installing just the latest version of acc (the backend) from Magisk?

kommserv, 2022-02-07:
I had not. Now I have tried it, but it did not work (details below).
And thanks for the suggestion, I appreciate it! If you or anyone else has any other suggestions, even if after a while, I would be grateful.

Downloaded the [latest zip](https://github.com/VR-25/acc/releases/download/v2022.2.3/acc_2022-02-03_14.16.05.zip) from https://github.com/VR-25/acc/releases, installed it using Magisk:
```
- Copying zip to temp directory
- Installing acc.zip

Advanced Charging Controller (ACC) v2022.2.3 (202202030)
Copyright 2017-2022, VR25
GPLv3+
(i) Installing in /data/adb/modules/acc/...
- Done

LINKS
- [Daily Job Scheduler](https://github.com/VR-25/djs)
- [Donate - Airtm, username: ivandro863auzqg](https://app.airtm.com/send-or-request/send)
- [Donate - Liberapay](https://liberapay.com/vr25)
- [Donate - Patreon](https://patreon.com/vr25)
- [Donate - PayPal or Credit/Debit Card](https://www.paypal.com/cgi-bin/webscr?cmd=_donations&business=iprj25@gmail.com&lc=US&item_name=VR25+is+creating+free+and+open+source+software.+Donate+to+suppport+their+work.&no_note=0&cn=&currency_code=USD&bn=PP-DonationsBF:btn_donateCC_LG.gif:NonHosted)
- [Facebook Page](https://fb.me/vr25xda)
- [Frontend - ACC App](https://github.com/MatteCarra/AccA/releases)
- [Frontend - ACC Settings](https://github.com/CrazyBoyFeng/AccSettings)
- [Must Read - How to Prolong Lithium Ion Batteries Lifespan](https://batteryuniversity.com/article/bu-808-how-to-prolong-lithium-based-batteries)
- [Telegram Channel](https://t.me/vr25_xda)
- [Telegram Group](https://t.me/acc_group)
- [Telegram Profile](https://t.me/vr25xda)
- [Upstream Repository](https://github.com/VR-25/acc)
- [XDA Thread](https://forum.xda-developers.com/apps/magisk/module-magic-charging-switch-cs-v2017-9-t3668427)

CHANGELOG
**v2022.2.3 (202202030)**
- Additional charging switches;
- Auto detect and blacklist unwritable charging control files and those that trigger unexpected reboots;
- Blacklisted troublesome charging switches;
- Fixed bg-dexopt-job wrapper causing long accd stop delay;
- Fixed typos in README.md;
- Fixed voltage "millivolts --exit" error;
- Improved charging status logic;
- Magisk updateJson support;
- Misc fixes & optimizations;
- Moved changelog from README.md to changelog.md;
- Removed README.html;
- Support for a new charging switch format: `ctrl_file1 <on_value|file> <off_value|file> ctrl_file2 <on_value|file> <off_value|file> ...`, `file` is where to get the on/off value from, e.g., `battery/charge_control_limit 0 battery/charge_control_limit_max`;
- Updated unexpected reboot troubleshooting guide.

**v2022.1.8 (202201080)**
- `acc -p` finds even more potential switches;
- Enhanced charging status detection;
- General fixes & optimizations;
- Improved idle mode support;
- New charging switches;
- Optimize system performance and battery utilization, by forcing `bg-dexopt-job` on daemon [re]start, if charging;
- Support for Qualcomm SnapDragon 8 Gen 1 devices, Nokia 2.2 and more;
- Updated documentation.

- **v2021.12.20 (202112200)**
- [accd, misc-functions]: prevent unwanted crashes related to `eval` and `set -eu`;
- [batt-info]: filter out the unreliable `POWER_SUPPLY_CHARGE_TYPE` property (note: this change makes AccA always display "unknown" charge type);
- [batt-info]: fixed current reading issue;
- [batt-info]: round current and voltage values to two decimal places;
- [ctrl-files]: added `battery/op_disable_charge 0 1` switch;
- [README]: updated troubleshooting section;
- General optimizations.

(i) Rebooting is unnecessary
- acc commands may require the /dev/.vr25/acc/ prefix (e.g., /dev/.vr25/acc/acc -v) until system is rebooted.
- Daemon started.
```

[... 44 lines cut ...]
(i) accd v2022.2.3 (202202030) is running (PID 12138)
```

Hm, well look at that! Without `/dev/.vr25/acc/.batt-interface.sh`, acc would not start at all, and neither would any front-end app. But simply creating the empty file seems to have resolved the problem?

Rebooted the tablet, opened terminal and checked ACC daemon again:
```
flounder:/ $ su
flounder:/ # accd,
/sbin/acc: /data/adb/vr25/acc/batt-interface.sh[97]: .: /dev/.vr25/acc/.batt-interface.sh: No such file or directory
(i) This may take a minute or so...
(i) /data/adb/vr25/acc-data/logs/acc-logs-flounder.tar.gz
```
Now this is very surprising! It seems some process has deleted the file has and now the ACC daemon is apparently broken again.

I went ahead and installed @CrazyBoyFeng's [AccSettings frontend app v2022.1.17-dev](https://github.com/CrazyBoyFeng/AccSettings/releases/tag/v2022.1.17-dev), and after granting it superuser it fails to start acc (screenshot):

[image]

Now what if we apply our "fix" and make sure to start the ACC daemon in the terminal?
Well, that seems to break the frontend app completely - it crashes and shows the Android pop-up "ACC Settings has stopped". Restarting the app does not help.

Just for completeness, I also tried to install [AccA from F-droid](https://f-droid.org/packages/mattecarra.accapp/) again, but it failed to start with the same error message as originally reported.

This is on the same system as originally reported, except we are now using Magisk 24.1 (so no built-in Magisk modules repo, but as you can tell I worked around that).

CrazyBoyFeng, 2022-02-07:
The last version of `AccSettings` had some issues with installing/updating acc, the fix will be released in the next version (but not soon). Until then, when using `AccSettings`, you can try to install it manually from the shell.

KeepBlueTomato, 2023-07-14:
> I had not. Now I have tried it, but it did not work (details below).
>
> And thanks for the suggestion, I appreciate it! If you or anyone else has any other suggestions, even if after a while, I would [...]

>just need open acc code but thank U AaA LoL

VR-25, 2023-07-15:
Have you tried the latest acc release?