MatteCarra/AccA #207: latest app and majisk module causing reboot

nutpantz, 2022-01-19:
Running the latest app and majisk module add of today. Charge limiting to 80% working, limit charge to start when below 70% not working.
Test charging switches causes reboot. Sometimes putting device on wireless charging causing reboot. Using galaxy s7 edge with lineage os Android 11 last updated 12-14-2021 with micro g

VR-25, 2022-01-23:
1. Plug the charger
2. Paste and run the following as root: `accd.; acc -x -t`
3. Share /sdcard/acc-*.log after the reboot

VR-25, 2022-01-23:
Troublesome charging switches trigger unexpected reboots on some systems.
To quickly identify these, plug the charger and run `acc -x -t`.
After the reboot, `/sdcard/acc-*.log` reveals the problematic switch.
It can then be removed from `ctrl-files.sh` or blacklisted.

nutpantz, 2022-01-28:
> 

Acc  -x  -t in a su terminal just shows the help info

VR-25, 2022-01-29:
Flash the latest acc snapshot from [here](https://forum.xda-developers.com/attachments/acc_2022-01-23_10-04-51-zip.5516117/).
Paste and run `pkill -9 accd; acc -x -t`.
Share `/sdcard/acc-*.log`.

nutpantz, 2022-01-29:
Failed to flash. Error 255

VR-25, 2022-01-29:
In recovery mode, you need to mount system first.
Did you try flashing via Magisk?

VR-25, 2022-01-29:
[acc_2022-01-29_18.16.59.zip](https://github.com/MatteCarra/AccA/files/7964201/acc_2022-01-29_18.16.59.zip)
See if this build also triggers a reboot.

nutpantz, 2022-01-29:
Flashing from magisk works. Recovery with system mounted still did not.
Info attached

nutpantz, 2022-01-29:
01-29 is also causing a reboot when hooked up to the charger and checking the settings in the app causes the app the freeze.
 
What's the best way to uninstall everything to start fresh?

VR-25, 2022-01-29:
That version has several charging control files blacklisted.
Do you mind going through steps 3 and 4 again to see what else should be blacklisted?

acc -U removes everything.

nutpantz, 2022-01-29:
1-29 That caused a reboot during the test
Here is the log. I love this app and oil do what i can help to make it better. Just ask.

VR-25, 2022-01-30:
You forgot the logs.
There's a new zip to share, but I would like to see your logs first.

nutpantz, 2022-01-30:
###Sat Jan 29 09:51:30 PST 2022###
versionCode=202201230
247: get_prop version /data/adb/vr25/acc/module.prop
129: sed -n 's|^version=||p' /data/adb/vr25/acc/module.prop
247: accVer=v2022.1.23 
248: get_prop versionCode /data/adb/vr25/acc/module.prop
129: sed -n 's|^versionCode=||p' /data/adb/vr25/acc/module.prop
248: accVerCode=202201230 
250: unset -f get_prop
253: misc_stuff -t
288: set -eu
289: 2>/dev/null 
289: mkdir -p /data/adb/vr25/acc-data
290: '[' -f /data/adb/vr25/acc-data/config.txt ']'
299: unset -f misc_stuff
254: tt -t '*/*'
344: eval $'case "-t" in\n    */*) return 0;;\n  esac'
347: return 1
347: >/dev/null 
347: 2>&1 
258: set +x
261: . /data/adb/vr25/acc-data/config.txt
1: configVerCode=202201010 
2: set -A capacity -- -1 60 70 75 false false
3: set -A temperature -- 40 60 90 65
4: set -A cooldownRatio --
5: cooldownCurrent='' 
6: set -A cooldownCustom --
7: set -A resetBattStats -- false false
8: set -A chargingSwitch --
9: set -A applyOnBoot --
10: set -A applyOnPlug --
11: set -A maxChargingCurrent --
12: set -A maxChargingVoltage --
13: language=en 
14: set -A runCmdOnPause --
15: ampFactor='' 
16: voltFactor='' 
17: set -A loopCmd --
18: prioritizeBattIdleMode=false 
19: currentWorkaround=false 
265: . /data/adb/vr25/acc/strings.sh
268: true
268: '[' -f /data/adb/vr25/acc/translations/en/strings.sh ']'
271: 2>/dev/null 
271: grep -q .. /data/adb/vr25/acc/translations/en/README.md
273: readMe=/data/adb/vr25/acc-data/README.md 
279: tt /system/bin/acc '*accd*'
344: eval $'case "/system/bin/acc" in\n    *accd*) return 0;;\n  esac'
347: return 1
443: shift
444: print_unplugged
64: echo '(!) Charger must be plugged to continue...'
446: >/dev/null 
446: daemon_ctrl stop
48: typeset 'isRunning=false'
50: <>/dev/.vr25/acc/acc.lock 
50: flock -n 0
66: false
71: print_not_running
16: echo '(i) accd is not running'
72: return 9
447: daemonWasUp=false 
449: . /data/adb/vr25/acc/acquire-lock.sh
2: id=acc 
3: 2>/dev/null 
3: set +o sh
4: 4<>/dev/.vr25/acc/acc.lock 
4: exec
5: <&4 
[... 2036 lines cut ...]
134: typeset 'failed=false'
135: set -A chargingSwitch -- ac/current_max 900000 0
137: flip_sw off
249: typeset 'on='
250: typeset 'off='
251: typeset 'off_='
252: typeset 'flip=off'
254: set -- ac/current_max 900000 0
255: '[' -f ac/current_max ']'
257: '[' -f ac/current_max ']'
258: echo 900000
258: sed 's/::/ /g'
258: on=900000 
259: '[' off '=' off ']'
260: tt 'ac/current_max 900000 0' 'battery/input_suspend?0?1?/proc/mtk_battery_cmd/en_power_path?1?1*' ']'
344: eval $'case "ac/current_max 900000 0" in\n    battery/input_suspend?0?1?/proc/mtk_battery_cmd/en_power_path?1?1*) return 0;;\n  esac'
347: return 1
261: '[' 0 '=' voltage_now ']'
267: echo 0
267: sed 's/::/ /g'
267: off=0 
269: >/dev/.vr25/acc/.curr 
269: cat battery/current_now
270: sleep 1
274: chmod 0644 ac/current_max

VR-25, 2022-01-30:
What's the output of `acc -ss:`?

nutpantz, 2022-01-30:
:/ $ su
:/ # acc -ss:

battery/batt_slate_mode 0 1
battery/hmt_ta_charge 1 0
battery/store_mode 0 1
max77854-fuelgauge/charge_enabled 1 0
ac/current_max 660000 0
battery/batt_tune_chg_limit_cur 1100 0
battery/batt_tune_fast_charge_current 2100 0
battery/batt_tune_intput_charge_current 1800 0
max77854-charger/current_max 660 0
usb/current_max 660000 0
wireless/current_max 660000 0
battery/batt_tune_float_voltage 43500 voltage_now
battery/fg_full_voltage 43500 voltage_now
max77854-charger/voltage_max 43500 voltage_now
0 mA
250 mA
350 mA
500 mA
3700 mV

:/ #

VR-25, 2022-01-30:
[acc_2022-01-30_20.38.01.zip](https://github.com/MatteCarra/AccA/files/7966763/acc_2022-01-30_20.38.01.zip)

Plug the charger, install the new build, run `pkill -9 accd; acc -t` and share `/data/adb/vr25/acc-data/logs/acc-t.log` after the reboot.

nutpantz, 2022-01-30:
Flashing that in magiskb caused a reboot and running the command caused a reboot. 
The requested log file was not in that directory, just a install and init log

VR-25, 2022-01-30:
1. Plug the charger and run `pkill -9 accd; acc -s
s="battery/batt_slate_mode 0 1 --"`
2. Reinstall the new build
3. Run `acc -t` and you should have the log file after the reboot

nutpantz, 2022-01-30:
No log file created. No reboot during flashing but it did reboot when i ran the command

nutpantz, 2022-01-30:
Still no log file

VR-25, 2022-01-31:
[acc_2022-01-31_19.07.13.zip](https://github.com/MatteCarra/AccA/files/7973268/acc_2022-01-31_19.07.13.zip)
This build will auto detect and blacklist the charging control files that are causing reboots.
Run `pkill -9 accd` before installing.

Once you run `acc -t` a few times, without reboots, share `/data/adb/vr-25/acc-data/logs/write.log`.

nutpantz, 2022-01-31:
#battery/batt_tune_chg_limit_cur
battery/batt_slate_mode
battery/charge_control_limit
battery/hmt_ta_charge
battery/store_mode
max77854-fuelgauge/charge_enabled

VR-25, 2022-01-31:
`#battery/batt_tune_chg_limit_cur` is the deal. The leading hashtag means it was behind the reboots and got blacklisted.
It's surprising, though, that this current control file would trigger a kernel panic. The kernel has major power management bugs.

Just to be sure, which issues are you experiencing now?
Share the output of `acc -t` and the log archive generated by `acc -le`.

nutpantz, 2022-01-31:
:/ $ su
:/ # acc -t

(!) Charger must be plugged to continue...
(i) This may take a minute or so...

(i) [battery/batt_slate_mode 0 1] works
- battIdleMode=false

(i) [battery/charge_control_limit 0 1] works
- battIdleMode=false

(!) [battery/hmt_ta_charge 1 0] won't work

(!) [battery/store_mode 0 1] won't work

(!) [max77854-fuelgauge/charge_enabled 1 0] won't work

:/ #

nutpantz, 2022-01-31:
[acc-logs-hero2ltebmc.tar.gz](https://github.com/MatteCarra/AccA/files/7973763/acc-logs-hero2ltebmc.tar.gz)

nutpantz, 2022-01-31:
I'm having
Reboots sometimes when placing the phone on a wireless charger or taking it off
Reboots when changing settings in the app
Sometimes the charging % limits from the app don't work
Sometimes the phone stops charging long before the % limit.
Usually a reboot gets the phone charging to the right limit for at least 12 hours or so. Reboots from charging or changing settings is mostly random.
Everything is with the fresh install of 2 different Android 11 lineage os roms

VR-25, 2022-02-01:
[acc_2022-02-01_15.36.09.zip](https://github.com/MatteCarra/AccA/files/7979238/acc_2022-02-01_15.36.09.zip)
I implemented workarounds for better compatibility with your system.

nutpantz, 2022-02-17:
I switched to Android 10 which has a few issues but nothing big enough for me to follow up on.
Android 11 on my mega 6.3 and acca works fine it seems. So it must be a device and rom issue. I'll follow up if there is a android 12 to test