MatteCarra/AccA #209: pixel 6 Android 12 acc not working

nutpantz, 2022-02-25:
Installing the service first time fails from app off fdroid. But works on retry.
Testing the charging switch works. 
Limits appear to be ignored.
Charges past limit, starts charging when above resume limit. Have not really tested anything else yet. Just got the pixel 6

nutpantz, 2022-02-26:
It appears that after i reach the % limits, the charging does not turn off, but cycles on and off every 2 seconds or so.

nutpantz, 2022-02-26:
:/ $ su
:/ #  acc -ss:

/sys/kernel/fast_charge/force_fast_charge 1 0
maxfg/offmode_charger 0 1
dc/current_max 687500 0
gcpm/constant_charge_current_max 1933090 0
pca9468-mains/current_max  0
wireless/current_max 700000 0
gcpm/constant_charge_voltage_max 4350000 voltage_now
gcpm/voltage_max 4350000 voltage_now
main-charger/voltage_max 4350000 voltage_now
0 mA
5 mA
10 mA
15 mA
50 mA
150 mA
250 mA
3700 mV
3800 mV
3850 mV
3920 mV

:/ #

nutpantz, 2022-03-01:
:/ $ su
:/ # pkill -9 accd; acc -x -t;

(!) Charger must be plugged to continue...
(i) This may take a minute or so...

(!) [/sys/kernel/fast_charge/force_fast_charge 1 0] won't work

(!) [maxfg/offmode_charger 0 1] won't work

(!) [dc/current_max 687500 0] won't work

(i) [gcpm/constant_charge_current_max 1933090 0] works
- battIdleMode=true

/system/bin/acc[594]: 3: parameter not set
15|:/ #

nutpantz, 2022-03-01:
I'm getting different results at different times.
:/ $ su
:/ # pkill -9 accd; acc -x -t;

(!) Charger must be plugged to continue...
(i) This may take a minute or so...

(i) [/sys/kernel/fast_charge/force_fast_charge 1 0] works
- battIdleMode=false

(i) [maxfg/offmode_charger 0 1] works
- battIdleMode=false

(i) Results may vary with different power supplies and conditions, as stated in README.md -- TROUBLESHOOTING.
To set charging switches, run acc -ss (wizard) or acc -s s="SWITCHES GO HERE --".
"battIdleMode" refers to the ability of running off the charger.

:/ #

This was run on the wireless charger which 8 user more often

nutpantz, 2022-03-02:
I don't know what i did.. But right now it's working i think.
But if you need any info feel free at ask. I'm sure i will have to install from scratch again at some point. And it would be nice if it was a simple install.

f1vefour, 2022-03-24:
> I don't know what i did.. But right now it's working i think. But if you need any info feel free at ask. I'm sure i will have to install from scratch again at some point. And it would be nice if it was a simple install.

I've noticed some strangeness with my Pixel 4 XL unless I leave the charge switch set to automatic even when selecting the correct switch manually.

nutpantz, 2022-05-09:
I don't know if its the latest version of acca or Android 12 adaptive charging. No matter the setting it stops at my maximum then very very slowly charges to 100% . I can't leave it to charge overnight

f1vefour, 2022-05-09:
> I don't know if its the latest version of acca or Android 12 adaptive charging. No matter the setting it stops at my maximum then very very slowly charges to 100% . I can't leave it to charge overnight

You have tried leaving Capacity Control on automatic and simply enabling Prioritize battery idle mode?

nutpantz, 2022-05-10:
Yes that fails badly as it seems the charge does not slow down after reaching max%
 I had issues with my s7 on Android 11 and 12 and had to go back to Android 10 before acc would work reliable again. Then i got my pixel 6 and have nothing but issues from a variety of vectors

Iey4iej3, 2022-07-01:
> unless specifically doing as described my 4XL also would eventually charge completely.

The same for me, but when the "Prioritize battery idle mode" is enabled, it will drain the battery, instead of keeping at max%. I think that this will also shorten the battery life by charging and discharging between the two percentages.

VR-25, 2022-07-05:
https://github.com/VR-25/acc/releases/download/v2022.7.4-dev/acc_v2022.7.4-dev_202207040_1927.zip

VR-25, 2022-07-07:
> It appears to still keep charging. Maybe 1% an hour past the maximum. Which i think is better then it was.

It's a charging switch problem.
Try another switch.

nutpantz, 2022-10-11:
Yes i hear tried that again to be sure, it seems to not slow the charging at all.

fillwithjoy1, 2023-08-30:
I'm using main-charger/current_max to control the charging and it does work. But I suppose as the name suggests, it only works for cables only and not for wireless charging