MatteCarra/AccA #210: Notice when daemon stop runnig

lavoie005, 2022-03-14:
Hi and thank you for your app.
Can you guys can add alarm for let us know when the daemon stop running.
I use pixel 6 and the daemon stop running.

VR-25, 2022-03-15:
The next acc release will have this feature.

Iey4iej3, 2022-05-31:
I install AccA from F-Droid. Sometimes, it stops after plugging in the USB to charge. I notice this since the charge level passed the limit. I will investigate what happened. Now I disabled battery optimization for AccA.

VR-25, 2022-06-04:
> I install AccA from F-Droid. Sometimes, it stops after plugging in the USB to charge. I notice this since the charge level passed the limit. I will investigate what happened. Now I disabled battery optimization for AccA.

This is irrelevant. AccA does not run in the background. The daemon is not tied to the app. It stops if all charging switches fail to respond. The next acc release will have better compatibility.

Iey4iej3, 2022-06-04:
> This is irrelevant. AccA does not run in the background. The daemon is not tied to the app. It stops if all charging switches fail to respond. The next acc release will have better compatibility.

However, I disabled battery optimization, and up until now, the issue does not occur again.

bunklung, 2022-10-26:
Pixel 7 and I have noticed twice so far where the daemon has stopped and my phone is 100% in the morning. I using 2021.8.31 and 1.0.35 (39).

closed 2026-10-30 by lavoie005