MatteCarra/AccA #212: AccA (F-Droid) freezes when changing Custom Settings

rgavioli, 2022-05-10:
Hello,

Not sure if this is the right place to report an issue with the AccA front end (F-Droid), so apologies in advance.

The back-end works absolutely fine (all commands), but when I try to use the F-Droid version (front-end) to apply a change on Custom Settings, the app freezes and I have to restart it.

I'm on Android 12 (Samsung S20), App version 1.0.35 and daemon v2022.2.22.1. I tried reinstalling it, clearing storage cache, clearing data and nothing seems to work.

I'll be glad to provide any logs you'd need.

Many thanks!

hanni2301, 2022-05-14:
actually i have the exact same problem:
A11
ACC daemon v2022.2.22.1
AccA 1.0.35

also uninstalled, installed, wiped, factory resetted the Pixel more then once.
The deamon is running fine. If you change the profile in the Profile-Tab you also don't get the coloured bar on the profile-banner and if you go back to the overview, there is only a loading circle. restarting the deamon does nothing. 
the only way to get the UI back to work is the kill the App AccA and relaunch it.

Something new to me is a working profile (via shelter) on my device. I don't know how to supply you more detailed informations.

Maozai-Teng, 2024-05-21:
I have the exact same problem too……QAQ